dusty-nv/jetson-inference #1599: failed to create videoOutput device when importing torch

Yiming0416, 2023-04-08:
Hi,

I am using Jetson nano 4GB with jetpack 4.6.1 and PyTorch 1.10. I built/installed jetson-inference from the source folllowing the instruction. I am able to run `my-detection.py` with my usb webcam. 

However, if I add `import torch` to `my-detection.py` (just adding the single line of code, no other changes), then I cannot run the code anymore, it gives

```
[OpenGL] failed to create X11 Window
Exception: jetson.utils -- failed to create videoOutput device
```

Yiming0416, 2023-04-08:
easier way to reproduce:
This code snippet works fine
```
from jetson_utils import videoSource, videoOutput
camera = videoSource("/dev/video0")      # '/dev/video0' for V4L2
display = videoOutput("display://0") # 'my_video.mp4' for file
while display.IsStreaming():
	img = camera.Capture()
	display.Render(img)
```

But this one is not working
```
import torch
from jetson_utils import videoSource, videoOutput
camera = videoSource("/dev/video0")      # '/dev/video0' for V4L2
display = videoOutput("display://0") # 'my_video.mp4' for file
while display.IsStreaming():
	img = camera.Capture()
	display.Render(img)
```
It gives the following error
```
[OpenGL] failed to create X11 Window.
Traceback (most recent call last):
  File "cuda_camera.py", line 5, in <module>
    display = videoOutput("display://0") # 'my_video.mp4' for file
Exception: jetson.utils -- failed to create videoOutput device
```

If I import torch after jetson_utils

```
from jetson_utils import videoSource, videoOutput
import torch
camera = videoSource("/dev/video0")      # '/dev/video0' for V4L2
display = videoOutput("display://0") # 'my_video.mp4' for file
while display.IsStreaming():
	img = camera.Capture()
	display.Render(img)
```

It gives the following error:

```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  43
  Current serial number in output stream:  42
```

dusty-nv, 2023-04-10:
I realize this is un-Pythonic, but can you try this way:

``` python
from jetson_utils import videoSource, videoOutput

camera = videoSource("/dev/video0")      # '/dev/video0' for V4L2
display = videoOutput("display://0") # 'my_video.mp4' for file

import torch
```

It would seem there is something in the torch / libtorch libraries that messes with the OpenGL extensions