dusty-nv/jetson-inference #1605: Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object

Boris7Britva, 2023-04-13:
Hi.

Jetson Nano 2Gb+JetPack 4.6.1 (clean installation)+CSI RPi cam v2

also tried 

Jetson Nano 2Gb+JetPack 4.5 (clean installation)CSI RPi cam v2

Camera streaming is fine on both.

There is an issue with detection networks from the examples folder, for example my_detection.py with SSD-Mobilenet-v2.

`boris@jp450:~/Desktop/AI/jetson-inference/python/examples$ python3 my-detection.py
jetson.inference -- detectNet loading custom model '(null)'

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 8.2.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - UFF  (extension '.uff')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +230, GPU +0, now: CPU 254, GPU 1817 (MiB)
[TRT]    [MemUsageSnapshot] Begin constructing builder kernel library: CPU 254 MiB, GPU 1846 MiB
[TRT]    [MemUsageSnapshot] End constructing builder kernel library: CPU 283 MiB, GPU 1876 MiB
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    found engine cache file /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.8201.GPU.FP16.engine
[TRT]    found model checksum /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.sha256sum
[TRT]    echo "$(cat /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.sha256sum) /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff" | sha256sum --check --status
[TRT]    model matched checksum /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.sha256sum
[TRT]    loading network plan from engine cache... /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.8201.GPU.FP16.engine
[TRT]    device GPU, loaded /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 288, GPU 1943 (MiB)
[TRT]    Loaded engine size: 33 MiB
[TRT]    Using cublas as a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +158, GPU -1, now: CPU 464, GPU 1934 (MiB)
[... 138 lines cut ...]

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[gstreamer] gstCamera -- onPreroll
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)30/1
[gstreamer] gstBufferManager -- recieved first frame, codec=raw format=nv12 width=1280 height=720 size=1008
[gstreamer] gstBufferManager -- recieved NVMM memory
[cuda]   allocated 4 ring buffers (8 bytes each, 32 bytes total)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
Traceback (most recent call last):
  File "my-detection.py", line 33, in <module>
    detections = net.Detect(img)
Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
GS`

dusty-nv, 2023-04-13:
Hi @Boris7Britva, can you try rebuilding with `-DENABLE_NVMM=off` cmake option?

``` bash
cd jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install
```

Boris7Britva, 2023-04-13:
Hi @dusty-nv , thank you for fast reply.

Rebuild done per your input - still same issue:

[log1.txt](https://github.com/dusty-nv/jetson-inference/files/11222894/log1.txt)

dusty-nv, 2023-04-13:
oh okay, after inspecting the logs again, that may not have been necessary.

Are you able to run video-viewer.py or detectnet.py okay?    I may need to update my-detection.py for something

Boris7Britva, 2023-04-15:
@dusty-nv is there anything else I can try?

dusty-nv, 2023-04-17:
@Boris7Britva I'm not sure why your camera freezes only when running detectnet, sorry about that.  To further debug it, are you able to process a video file with detectnet okay?    https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#processing-video-files

Also, have you tried running the jetson-inference docker container to see if you get different results?   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

dusty-nv, 2023-04-17:
@Boris7Britva thanks, unfortunately I'm unable to reproduce this - can you try commenting out these two lines from detectnet.py:

https://github.com/dusty-nv/jetson-inference/blob/56604b07da34bdf32c02e1201ab99c413b47bc85/python/examples/detectnet.py#L90

then make sure you run the detectnet.py that has your changed modifications (or rename it to something else)

Boris7Britva, 2023-04-18:
@dusty-nv, please see the log:

[log_detectnet_edited.txt](https://github.com/dusty-nv/jetson-inference/files/11257696/log_detectnet_edited.txt)

dusty-nv, 2023-04-18:
@Boris7Britva everything looks normal in that log except the video seems to stop early.   I wonder if the video is corrupted?

Can you try running this example?  https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-console-2.md#processing-a-video

dusty-nv, 2023-04-20:
@Boris7Britva I still see a lot of errors in your logs about nvbuf utils - are you sure that you re-build/re-installed with `-DENABLE_NVMM=off`?

``` bash
cd jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install
```

(this should be done outside of container and you shouldn't use container)

Boris7Britva, 2023-04-20:
@dusty-nv Sorry, I reinstalled the image yesterday and didn't apply that option.

Attached log after disabling NVMM

[log_jellyfish_nvmm_off.txt](https://github.com/dusty-nv/jetson-inference/files/11288482/log_jellyfish_nvmm_off.txt)

dusty-nv, 2023-04-20:
OK, I believe the issue now is that you are trying to save the video to a directory that doesn't exist in your current path:

``` bash
~/Downloads/AI/jetson-inference/python/examples$ ./imagenet.py --network=resnet-18 jellyfish.mkv images/test/jellyfish_resnet18.mkv
```

(there is no `images/test` directory under `jetson-inference/python/examples` to save to)
Instead, just save `jellyfish_resnet18.mkv` to the current directory - try just running it as:

```
~/Downloads/AI/jetson-inference/python/examples$ ./imagenet.py --network=resnet-18 jellyfish.mkv jellyfish_resnet18.mkv
```

Boris7Britva, 2023-04-20:
Ok. This one worked. 

I've bought a new Rpi cam v2.1 to make sure it's not camera issue.

Can you please explain what's wrong for this simple script?

[log_jetson_csi.txt](https://github.com/dusty-nv/jetson-inference/files/11288988/log_jetson_csi.txt)
[jetson_csi.py.txt](https://github.com/dusty-nv/jetson-inference/files/11288993/jetson_csi.py.txt)

mmhzlrj, 2023-04-27:
Hi@Boris7Britva @dusty-nv . I am a beginner that met the same problem on jetson.inference.detectNet().
I was running these lines,and the csi camera was running just fine:
import jetson.utils
input = jetson.utils.videoSource("csi://0")
output = jetson.utils.videoOutput("display://0")

while output.IsStreaming():
    img = input.Capture()
    output.Render(img)

But after I put detectNet in to the lines, it showed the same problem that @Boris7Britva got:

import jetson.utils
input = jetson.utils.videoSource("csi://0")
output = jetson.utils.videoOutput("display://0")
import jetson.inference
net = jetson.inference.detectNet("ssd-mobilenet-v2",threshold=0.5)
 
while output.IsStreaming():
    img = input.Capture()
    detections = net.Detect(img)
    output.Render(img)
    output.SetStatus("Performance:{:.0}FPS".format(net.GetNetworkFPS()))

Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object

[log.txt](https://github.com/dusty-nv/jetson-inference/files/11339483/log.txt)
I have already tried  re-build/re-installed with -DENABLE_NVMM=off and it still showed the same exception.
I am using Jetson Nano 2Gb+ JetPack 4.6.3 with L4T 32.7.3 and IMX219 CSI camera.

Can you help me to fix this please? Thank you so much.

mmhzlrj, 2023-04-27:
I tried to use another TF card to reinstall the image and run  the script again. Still had the same exception, but the log looked differently.
[log.txt](https://github.com/dusty-nv/jetson-inference/files/11344084/log.txt)

mmhzlrj, 2023-04-28:
Thank you @dusty-nv . Firstly, I checked Line 65 in detectnet.py and add that `if img is None: `like this:
`while True:
    # capture the next image
    img_input = input.Capture()

    if img_input is None: # timeout
        continue
        
    img = input.Capture()

    if img is None: # timeout
        continue  `
And  I noticed that we got ` if img_input is None:` as well.

Secondly, I ran `nvgstcapture-1.0` on terminal and the first time the video stream or window didn't show up. And then I turned off Jetson nano and check the connection of my CSI camera and ran `nvgstcapture-1.0` again. Then I saw the video is streaming. But when I ran my Script:
`import jetson_utils
input = jetson_utils.videoSource("csi://0")
output = jetson_utils.videoOutput("display://0")

import jetson_inference
net = jetson_inference.detectNet("ssd-mobilenet-v2",threshold=0.5)

while output.IsStreaming():
	img = input.Capture()
	detections = net.Detect(img)
	output.Render(img)`
It still shows `gstCamera::Capture() -- a timeout occurred waiting for the next image buffer` and ` File "1.py", line 10, in <module>
    detections = net.Detect(img)
Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object
`
[log.txt](https://github.com/dusty-nv/jetson-inference/files/11349531/log.txt)
Thanks for your help@dusty-nv

mmhzlrj, 2023-04-28:
If I try usb camera instead of csi camera, the script ran well. I wonder which part went wrong.

dusty-nv, 2023-04-28:
> It still shows
> `gstCamera::Capture() -- a timeout occurred waiting for the next image buffer` and
> ` File "1.py", line 10, in <module> detections = net.Detect(img) Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object`

@mmhzlrj modify your script to be like this:

``` python
while output.IsStreaming():
	img = input.Capture()
	if img is None:
		continue
	detections = net.Detect(img)
	output.Render(img)
```

mmhzlrj, 2023-04-29:
```
gstCamera::Capture() -- a timeout occurred waiting for the next image buffer and
 File "1.py", line 10, in <module> detections = net.Detect(img) Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object
```
Problem solved. Thank you so much @dusty-nv

mathiasstreet, 2024-03-23:
Hi @dusty-nv ,

I am also currently having this problem whilst trying to run a modified version of my-detection.py I have made. I am currently using a CLI therefore, I removed all instances of the display object.

Heres the Code and Log:

`from jetson_inference import detectNet
from jetson_utils import videoSource

net = detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = videoSource("csi://0")      # '/dev/video0' for V4L2

while True:
    img = camera.Capture()

    detections = net.Detect(img)
    
    print(detections)`

[log.txt](https://github.com/dusty-nv/jetson-inference/files/14732084/log.txt)

Note: I have denabled NVMM using the following commands:

cd jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install

Any help would be appreciated,

Thanks,

Mathias