dusty-nv/jetson-inference #1606: /usr/bin/ld: cannot find -lvpi when compiling segnet.cpp by myself

felixshing, 2023-04-15:
I have successfully built the project from the source based on this [tutorial:](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) 

Now, I want to build the segnet.cpp by myself for future need. I only changed the way the headers are defined:
```
#include "jetson-utils/videoSource.h"
#include "jetson-utils/videoOutput.h"

#include "jetson-utils/cudaOverlay.h"
#include "jetson-utils/cudaMappedMemory.h"

#include "jetson-inference/segNet.h"
```

The followings are my CMakeLists.txt

```
cmake_minimum_required(VERSION 2.8)

# declare my-recognition project
project(segnet)

find_package(jetson-utils REQUIRED)
find_package(jetson-inference REQUIRED)

# CUDA is required
#find_package(CUDA)
find_package(CUDA  REQUIRED)
include_directories("${CUDA_INCLUDE_DIRS}")
# add directory for libnvbuf-utils to program
link_directories(/usr/lib/aarch64-linux-gnu/tegra)

# compile the my-recognition program
add_executable(segnet segnet.cpp)
# link my-recognition to jetson-inference library
target_link_libraries(segnet jetson-inference)
install(TARGETS segnet DESTINATION bin)
```

When I run make -j$(nproc), I get this error:
`/usr/bin/ld: cannot find -lvpi`

I try to add
`link_directories(/opt/nvidia/vpi2/lib/aarch64-linux-gnu)`. But it does not work. How can I solve this issue?

felixshing, 2023-04-15:
Solve this problem by adding

`include_directories(/usr/include/gstreamer-1.0 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/include /usr/include/glib-2.0 /usr/include/libxml2 /usr/lib/aarch64-linux-gnu/glib-2.0/include/)`

and 

`target_link_libraries(Meta-segnet jetson-inference ${OpenCV_LIBRARIES} jetson-utils)`

closed 2023-04-15 by felixshing

AhmedL99, 2024-02-07:
Hello @felixshing , 
I am working with `detectnet` module. I am still having the same issues even after updating the CMakeLists file after your suggestions.

AhmedL99, 2024-02-08:
Hello @dusty-nv, 
Is there anyone who can help here? 
Thanks

Qengineering, 2024-05-31:
Add `find_package(VPI 2.0)` to your CMakeLists.txt