dusty-nv/jetson-inference #1607: How to get the results from segnet.cpp

felixshing, 2023-04-15:
Hello, I am trying to run segnet to segment the person (which is my target object) from the background, but I don't know how to get this result. Can anyone help me with that please?

dusty-nv, 2023-04-17:
@felixshing you could try using a segmentation model that has people and just ignore all the other classes, or you may want to try the backgroundNet, which is explicitly for foreground/background segmentation:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/backgroundnet.md

felixshing, 2023-04-19:
@dusty-nv Thank you very much for your reply. I would like to ask how and where can I modify the detected classes? Moreover, I am unsure what results I should get for my next step and how I can get it (for both segnet and background net)

I am currently using zed 2i depth camera to capture the rgb-d information. My goal is to extract my target (person) from the background and synthesize it to the point cloud to generate volumetric content. How could I utilize segnnt or background net to implement this? Specifically, what is the format of the output we should get and how could I use the output for my next step?

Thank you for your time.