dusty-nv/jetson-inference #1610: docker/run.sh not work, notthing show up

knowlessthanenough, 2023-04-19:
after i run docker/run.sh nothing show up. cmd show:
root@hkuit155-desktop:/home/hkuit155/jetson-inference# docker/run.sh
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.0.1
CONTAINER:      dustynv/jetson-inference:r34.0.1
DATA_VOLUME:    --volume /home/hkuit155/jetson-inference/data:/jetson-inference/data --volume /home/hkuit155/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/hkuit155/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/hkuit155/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/hkuit155/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:    
USER_COMMAND:   
V4L2_DEVICES:   
localuser:root being added to access control list
DISPLAY_DEVICE: -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix
docker/run.sh: line 179: /tmp/nv_jetson_model: Permission denied
sudo: docker: command not found

dusty-nv, 2023-04-19:
> sudo: docker: command not found

Hi @knowlessthanenough, do you have docker installed?   It should have been installed when the board was flashed.   You are also running a Developer Preview version of JetPack and you might want to upgrade it to get a fresh start.

knowlessthanenough, 2023-04-24:
i am so sorry about that. But after i install all the thing. i still got error.

hkuit155@hkuit155-desktop:~/jetson-inference$ cat /etc/nv_tegra_release
# R34 (release), REVISION: 1.1, GCID: 30414990, BOARD: t186ref, EABI: aarch64, DATE: Tue May 17 04:20:55 UTC 2022

hkuit155@hkuit155-desktop:~/jetson-inference$ sudo apt-cache show nvidia-jetpack
[sudo] password for hkuit155: 
Package: nvidia-jetpack
Version: 5.0.1-b118
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 5.0.1-b118), nvidia-jetpack-dev (= 5.0.1-b118)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.0.1-b118_arm64.deb
Size: 29304
SHA256: 78f182886cc85b6467f37e32553ee79adcf4aaf6b68031fe948f33cc8e008f92
SHA1: 3371de1f109f313379ed19f41a860fb60a555c86
MD5sum: 51cf5d858d2e533f063f3a7467bed6da
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

hkuit155@hkuit155-desktop:~/jetson-inference$ docker/run.sh
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.1.1
CONTAINER:      dustynv/jetson-inference:r34.1.1
DATA_VOLUME:    --volume /home/hkuit155/jetson-inference/data:/jetson-inference/data --volume /home/hkuit155/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/hkuit155/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/hkuit155/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/hkuit155/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:    
USER_COMMAND:   
V4L2_DEVICES:   
localuser:root being added to access control list
DISPLAY_DEVICE: -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix
docker: Error response from daemon: Unknown runtime specified nvidia.
See 'docker run --help'.