dusty-nv/jetson-inference #1611: cmake fail while build

knowlessthanenough, 2023-04-19:


knowlessthanenough, 2023-04-19:
The reason is that the gcc and g++ compilation environments are not installed. The installation method is very simple, just one line of command. Enter sudo apt-get install build-essential in the terminal to install it. In order to check whether it is installed, you can enter gcc --version, g++ --version in sequence to check the version numbers of gcc and g++ installed. If the following information is entered, the installation is successful.

Copyright statement: This is an original article by CSDN blogger "zeeq_", following the CC 4.0 BY-SA copyright agreement, please attach the original source link and this statement for reprinting.
Original link: https://blog.csdn.net/weixin_44120025/article/details/118769262

knowlessthanenough, 2023-04-19:
after solve his problem i get another error.
this time it show the python install ui and able to run, however i get this at the end.

[jetson-inference]  Package selection status:  0
[jetson-inference]  Packages selected for download:  1
[jetson-inference]  Downloading PyTorch v1.12.0 (Python 3.8)...
[jetson-inference]  Checking for 'python3-pip' deb package...installed
[jetson-inference]  Checking for 'qtbase5-dev' deb package...installed
[jetson-inference]  Checking for 'libjpeg-dev' deb package...installed
[jetson-inference]  Checking for 'zlib1g-dev' deb package...installed
[jetson-inference]  Checking for 'libopenblas-base' deb package...installed
[jetson-inference]  Checking for 'libopenmpi-dev' deb package...installed
[jetson-inference]  Checking for 'libomp-dev' deb package...installed
[jetson-inference]  Checking for 'ninja-build' deb package...installed
Requirement already satisfied: Cython in /home/hkuit155/.local/lib/python3.8/site-packages (0.29.34)
User install by explicit request
Created temporary directory: /tmp/pip-ephem-wheel-cache-qjq16akq
Created temporary directory: /tmp/pip-req-tracker-u0nqa7vx
Initialized build tracking at /tmp/pip-req-tracker-u0nqa7vx
Created build tracker: /tmp/pip-req-tracker-u0nqa7vx
Entered build tracker: /tmp/pip-req-tracker-u0nqa7vx
Created temporary directory: /tmp/pip-install-xfj6elog
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (1.17.4)
Cleaning up...
Removed build tracker: '/tmp/pip-req-tracker-u0nqa7vx'
User install by explicit request
Created temporary directory: /tmp/pip-ephem-wheel-cache-4z28386o
Created temporary directory: /tmp/pip-req-tracker-y8nmitkv
Initialized build tracking at /tmp/pip-req-tracker-y8nmitkv
Created build tracker: /tmp/pip-req-tracker-y8nmitkv
Entered build tracker: /tmp/pip-req-tracker-y8nmitkv
Created temporary directory: /tmp/pip-install-wpwgf3wi
Requirement already satisfied: tensorboard in /home/hkuit155/.local/lib/python3.8/site-packages (2.12.2)
Requirement already satisfied: wheel>=0.26 in /usr/lib/python3/dist-packages (from tensorboard) (0.34.2)
Requirement already satisfied: markdown>=2.6.8 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (3.4.3)
Requirement already satisfied: numpy>=1.12.0 in /usr/lib/python3/dist-packages (from tensorboard) (1.17.4)
Requirement already satisfied: requests<3,>=2.21.0 in /usr/lib/python3/dist-packages (from tensorboard) (2.22.0)
Requirement already satisfied: absl-py>=0.4 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (1.4.0)
Requirement already satisfied: grpcio>=1.48.2 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (1.54.0)
Requirement already satisfied: setuptools>=41.0.0 in /usr/lib/python3/dist-packages (from tensorboard) (45.2.0)
Requirement already satisfied: google-auth-oauthlib<1.1,>=0.5 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (1.0.0)
Requirement already satisfied: tensorboard-data-server<0.8.0,>=0.7.0 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (0.7.0)
Requirement already satisfied: tensorboard-plugin-wit>=1.6.0 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (1.8.1)
Requirement already satisfied: protobuf>=3.19.6 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (4.22.3)
Requirement already satisfied: werkzeug>=1.0.1 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (2.2.3)
Requirement already satisfied: google-auth<3,>=1.6.3 in /home/hkuit155/.local/lib/python3.8/site-packages (from tensorboard) (2.17.3)
Requirement already satisfied: importlib-metadata>=4.4; python_version < "3.10" in /home/hkuit155/.local/lib/python3.8/site-packages (from markdown>=2.6.8->tensorboard) (6.5.0)
Requirement already satisfied: requests-oauthlib>=0.7.0 in /home/hkuit155/.local/lib/python3.8/site-packages (from google-auth-oauthlib<1.1,>=0.5->tensorboard) (1.3.1)
Requirement already satisfied: MarkupSafe>=2.1.1 in /home/hkuit155/.local/lib/python3.8/site-packages (from werkzeug>=1.0.1->tensorboard) (2.1.2)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3/dist-packages (from google-auth<3,>=1.6.3->tensorboard) (1.14.0)
Requirement already satisfied: cachetools<6.0,>=2.0.0 in /home/hkuit155/.local/lib/python3.8/site-packages (from google-auth<3,>=1.6.3->tensorboard) (5.3.0)
Requirement already satisfied: rsa<5,>=3.1.4; python_version >= "3.6" in /home/hkuit155/.local/lib/python3.8/site-packages (from google-auth<3,>=1.6.3->tensorboard) (4.9)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /home/hkuit155/.local/lib/python3.8/site-packages (from google-auth<3,>=1.6.3->tensorboard) (0.2.8)
Requirement already satisfied: zipp>=0.5 in /home/hkuit155/.local/lib/python3.8/site-packages (from importlib-metadata>=4.4; python_version < "3.10"->markdown>=2.6.8->tensorboard) (3.15.0)
Requirement already satisfied: oauthlib>=3.0.0 in /usr/lib/python3/dist-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib<1.1,>=0.5->tensorboard) (3.1.0)
Requirement already satisfied: pyasn1>=0.1.3 in /home/hkuit155/.local/lib/python3.8/site-packages (from rsa<5,>=3.1.4; python_version >= "3.6"->google-auth<3,>=1.6.3->tensorboard) (0.4.8)
Cleaning up...
Removed build tracker: '/tmp/pip-req-tracker-y8nmitkv'
torch-1.12.0a0+2c91 100%[===================>] 220.68M  67.1MB/s    in 3.3s    
[... cut ...]
   used as include directory in directory /home/hkuit155/jetson-inference/python/bindings
   used as include directory in directory /home/hkuit155/jetson-inference/python/bindings
   used as include directory in directory /home/hkuit155/jetson-inference/python/bindings
   used as include directory in directory /home/hkuit155/jetson-inference/python/bindings
   used as include directory in directory /home/hkuit155/jetson-inference/python/bindings

-- Configuring incomplete, errors occurred!
See also "/home/hkuit155/jetson-inference/build/CMakeFiles/CMakeOutput.log".

dusty-nv, 2023-04-19:
> CUDA_TOOLKIT_ROOT_DIR not found or specified
> -- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)

Hi @knowlessthanenough, do you have CUDA Toolkit installed?  It should have already been installed when you flashed the board.   It seems from your other topic that you might want to re-flash with the latest version of JetPack.

knowlessthanenough, 2023-04-24:
i reinstall the jetpack but still got same problem.
hkuit155@hkuit155-desktop:~/jetson-inference/build$ sudo apt-cache show nvidia-jetpack
Package: nvidia-jetpack
Version: 5.0.1-b118
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 5.0.1-b118), nvidia-jetpack-dev (= 5.0.1-b118)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.0.1-b118_arm64.deb
Size: 29304
SHA256: 78f182886cc85b6467f37e32553ee79adcf4aaf6b68031fe948f33cc8e008f92
SHA1: 3371de1f109f313379ed19f41a860fb60a555c86
MD5sum: 51cf5d858d2e533f063f3a7467bed6da
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

is there any other place i may do it wrong?

knowlessthanenough, 2023-04-24:
Here's the cmake output log. I didn't find any error inside and I feel confused about what to do. Do you have any ideas? Thank.
[CMakeOutput.log](https://github.com/dusty-nv/jetson-inference/files/11306227/CMakeOutput.log)

closed 2023-04-27 by dusty-nv