dusty-nv/jetson-inference #1612: docker/run.sh  issue

drvrh, 2023-04-19:
Hello, 

I run docker/run.sh command and I get:

```
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.7.1
CONTAINER:      dustynv/jetson-inference:r32.7.1
DATA_VOLUME:    --volume /home/jetson16/jetson-inference/data:/jetson-inference/data --volume /home/jetson16/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/jetson16/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/jetson16/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/jetson16/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:    
USER_COMMAND:   
V4L2_DEVICES:   
localuser:root being added to access control list
DISPLAY_DEVICE: -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix
root@jetson16:/jetson-inference# 
```

What is wrong?

dusty-nv, 2023-04-19:
Hi @drvrh, I don't see anything as being wrong - it has started the container and given you the command-line prompt into it:

```
root@jetson16:/jetson-inference# 
```

You can then type commands in there and they will run inside the container.

dusty-nv, 2023-04-20:
Hi @drvrh, are you able to `ls /home/jetson16/projects/opencvtest/` ?

The path will be mounted to `/home/jetson16/projects/opencvtest/` inside the container (the second part of the `--volume` command after the `:`)