dusty-nv/jetson-inference #1613: can i modify detectnet for "real time detectnet streaming without display out"

EvaBarret1, 2023-04-20:
real time detectnet streaming without display out

dusty-nv, 2023-04-20:
@EvaBarret1 yes, you can run it with RTP/RTSP/WebRTC as the output:    https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#output-streams

You can simply specify these options on the command-line when running detectnet/detectnet.py

EvaBarret1, 2023-04-24:
and how to remove box fill in detectnet?

dusty-nv, 2023-04-24:
> Render(img) displays the image from the camera on the screen, I can't avoid it in any way

Hi @EvaBarret1, use the `--headless` option when you create your videoOutput interface and that will disable the GUI window:

``` python
output = videoOutput("rtsp://192.168.1.35:5000/live", argv='--headless')
```

> and how to change color box in detectnet?
Create a `colors.txt` file in your model's directory, and load it like this:

``` python
detectNet(model="flower4.4/ssd-mobilenet.onnx", labels="flower4/labels.txt", colors="flower4/colors.txt",
input_blob="input_0", output_cvg="scores", output_bbox="boxes",
threshold=args.threshold)
```

Your colors.txt file should have one line per class, with each line having `R G B` or `R G B A` color (between 0-255).  For example:

```
0 255 0
127 127 0
100 100 100
```

EvaBarret1, 2023-04-25:
thank you very much, you are the best

closed 2023-04-27 by dusty-nv

EvaBarret1, 2023-05-02:
Hello can i modifi bitrate in detectnet for v4l2 camera from4000000 to 1000000