dusty-nv/jetson-inference #1615: detectnet-tracking: 'jetson.inference.detectNet' object has no attribute 'SetTrackingEnabled'

ramzesko, 2023-04-21:
Hi Dustin,

I'm trying to follow your [tutorial] (https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-tracking.md) to add tracking feature to my app. I updated the repo on my jetson nano board when figured out that is quite fresh piece in your project but still cannot initialize detectNet with tracking enabled:

    Traceback (most recent call last):
    File "main_overlay_wide_mod_w_obj_tracking_1.py", line 68, in <module>
    tracker.SetTrackingEnabled(True)
    AttributeError: 'jetson.inference.detectNet' object has no attribute 'SetTrackingEnabled'

I have checked my _PyDetectNet.cpp_ file in bindings catalogue and the definitions of tracking suff are present there, for ex.:
       
    // SetTrackingEnabled
       static PyObject* PyDetectNet_SetTrackingEnabled( PyDetectNet_Object* self, PyObject* args )
       {
	if( !self || !self->net )
	{
		PyErr_SetString(PyExc_Exception, LOG_PY_INFERENCE "detectNet invalid object instance");
		return NULL;
	}	
	int enabled = 0;
 
	if( !PyArg_ParseTuple(args, "p", &enabled) )
		return NULL;

	objectTracker* tracker = self->net->GetTracker();

	if( tracker != NULL )
	{
		tracker->SetEnabled((bool)enabled);
		Py_RETURN_NONE;
	}
	
	if( !enabled )
		Py_RETURN_NONE;
	
	self->net->SetTracker(objectTracker::Create(objectTracker::IOU));
	
	Py_RETURN_NONE;
    }

Am I doing something wrong or the feature is not ready to use with python yet?

dusty-nv, 2023-04-21:
Hi @ramzesko, yes it should be there - can you try rebuilding/reinstalling:

``` bash
cd jetson-inference/build
cmake ../
make
sudo make install
```

If you are using the container, which version of L4T are you running?   I may not have updated the container for your L4T version, or you may have to explicitly pull the updated one.  Or you could rebuild the container yourself.

ramzesko, 2023-04-24:
Thanks @dusty-nv for your advice. Checked the environment once again and found out that the image was not updated. After pulling the right image it works like a charm.

closed 2023-04-27 by dusty-nv