dusty-nv/jetson-inference #1616: my-recognition$ make has compile error on ORIN JP 5.0.2

cosmos-star, 2023-04-21:
Followed implementation instructions from imagenet-example-2.md. During $make I get:
`
steven@development-orin:~/Projects/my-recognition$ make
[ 50%] Building CXX object CMakeFiles/my-recognition.dir/my-recognition.cpp.o
/home/steven/Projects/my-recognition/my-recognition.cpp: In function 'int main(int, char**)':
/home/steven/Projects/my-recognition/my-recognition.cpp:32:45: error: 'RESNET_18'  is not a member of 'imageNet'
   32 |  imageNet* net = imageNet::Create(imageNet::RESNET_18);
      |                                             ^~~~~~~~~
make[2]: *** [CMakeFiles/my-recognition.dir/build.make:76: CMakeFiles/my-recognition.dir/my-recognition.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/my-recognition.dir/all] Error 2
make: *** [Makefile:91: all] Error 2`

I deleted jetson-inference and re-installed. Made sure GOOGLENET & RESNET_18 were loaded  via $ ./download-models.sh
Still have issue. Recently upgraded to ORIN from Jetson NANO. No problem on NANO.

**Any ideas or hints?**

Note: I did have issue with 3 programs during inference installation. I fixed them so they would compile and I could move forward:

This errors appeared during the 'jetson-inference/build$ make' step.
`
1. PyNumpy.cpp

[ 91%] Building CXX object utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/PyNumpy.cpp.o
/home/steven/jetson-inference/utils/python/bindings/PyNumpy.cpp: In function ‘PyObject* PyNumpy_ImportNumpy()’:
/home/steven/jetson-inference/utils/python/bindings/PyNumpy.cpp:313:1: warning: control reaches end of non-void function [                                                              -Wreturn-type]
  313 | }
      | ^
------FIX-----------------------------------------------------------------------------------------------------------
308 // Initialize NumPy
309 PyMODINIT_FUNC PyNumpy_ImportNumpy()
310 {
311	import_array();
312	//import_ufunc();    // only needed if using ufunctions
313	return NULL;         // SPARKS 2023-04-20 Added this line to compile
314 }

2. tensorNet.cpp

[ 49%] Building CXX object CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o
/home/steven/jetson-inference/c/tensorNet.cpp: In member function ‘bool tensorNet::LoadNetwork(const char*, const char*, c                                                              onst char*, const std::vector<std::__cxx11::basic_string<char> >&, const std::vector<nvinfer1::Dims3>&, const std::vector<                                                              std::__cxx11::basic_string<char> >&, uint32_t, precisionType, deviceType, bool, nvinfer1::IInt8Calibrator*, cudaStream_t)’                                                              :
/home/steven/jetson-inference/c/tensorNet.cpp:1216:25: warning: ‘.calibration’ directive writing 12 bytes into a region of                                                               size between 1 and 4096 [-Wformat-overflow=]
 1216 |  sprintf(cache_path, "%s.calibration", cache_prefix);
      |                         ^~~~~~~~~~~~
/home/steven/jetson-inference/c/tensorNet.cpp:1216:9: note: ‘sprintf’ output between 13 and 4108 bytes into a destination                                                               of size 4096
 1216 |  sprintf(cache_path, "%s.calibration", cache_prefix);
      |  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/steven/jetson-inference/c/tensorNet.cpp:1222:25: warning: ‘.engine’ directive writing 7 bytes into a region of size                                                               between 1 and 4096 [-Wformat-overflow=]
 1222 |  sprintf(cache_path, "%s.engine", cache_prefix);
      |                         ^~~~~~~
/home/steven/jetson-inference/c/tensorNet.cpp:1222:9: note: ‘sprintf’ output between 8 and 4103 bytes into a destination o                                                              f size 4096
 1222 |  sprintf(cache_path, "%s.engine", cache_prefix);
      |  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-------FIX---------------------------------------------------------------------------------------------------------
1212 char cache_prefix[PATH_MAX]; 
1213 char cache_path[PATH_MAX + 20];   // SPARKS 2023-04-20Add 20 to length to compile

3. captureWindow.cpp

[ 75%] Building CXX object tools/camera-capture/CMakeFiles/camera-capture.dir/captureWindow.cpp.o
/home/steven/jetson-inference/tools/camera-capture/captureWindow.cpp: In member function ‘glWidget* CaptureWindow::GetWidg                                                              et(int) const’:
/home/steven/jetson-inference/tools/camera-capture/captureWindow.cpp:222:1: warning: no return statement in function retur                                                              ning non-void [-Wreturn-type]
  222 | }
      | ^
---------FIX-----------------------------------------------------------------------------------------------------
218 // GetWidget
219 glWidget* CaptureWindow::GetWidget( int index ) const
220 {
221	return display->GetWidget(index); // SPARKS 2023-04-20 Added return to compaile
222 }
`

dusty-nv, 2023-04-21:
sorry @cosmos-star - those warnings are unrelated.   Just fixed the my-recognition build in https://github.com/dusty-nv/jetson-inference/commit/6e3cd10b65ed0cbc9457aa6e513f355de75a3dc0 - try pulling the latest or updating your local copy.

cosmos-star, 2023-04-21:
That fixed the issues. Thanks for being so speedy!

closed 2023-04-27 by dusty-nv