dusty-nv/jetson-inference #1619: cant use webrtc,how to fix it?

tchy, 2023-04-24:
when run `video-viewer csi://0 webrtc://@:8554/output` and open `https://192.168.0.108:8554`, happen the error in this:

```
[webrtc] HTTPS 192.168.0.107 GET '/'
[webrtc] /
video-viewer:  captured 425 frames (1280x720)
video-viewer:  captured 450 frames (1280x720)
[webrtc] HTTPS 192.168.0.107 GET '/favicon.ico'
[webrtc] /favicon.ico
[webrtc] websocket /output -- new connection opened by 192.168.0.107 (peer_id=0)
[webrtc] new WebRTC peer connecting (192.168.0.107, peer_id=0)

(video-viewer:2138): GLib-GObject-WARNING **: 16:42:18.021: g_object_set_is_valid_property: object class 'GstWebRTCBin' has no property named 'latency'
**
ERROR:/home/jetson/jetson-inference/utils/codec/gstEncoder.cpp:866:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/home/jetson/jetson-inference/utils/codec/gstEncoder.cpp:866:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Aborted (core dumped)
```

dusty-nv, 2023-04-24:
Thanks @tchy, which Jetson and version of JetPack-L4T are you running?  (you can check this with `/etc/cat/nv_tegra_release`)

Also, are you using the jetson-inference container, or did you build it from source?

tchy, 2023-04-25:
> Thanks @tchy, which Jetson and version of JetPack-L4T are you running? (you can check this with `/etc/cat/nv_tegra_release`)
> 
> Also, are you using the jetson-inference container, or did you build it from source?

Thanks @dusty-nv ，I build it from source。I don't know JetPack-L4T whether works，I input `/etc/cat/nv_tegra_release`，it say “No such file or directory”。But，I other videoout is work, such rtsp.

dusty-nv, 2023-04-25:
> I don't know JetPack-L4T whether works，I input /etc/cat/nv_tegra_release，it say “No such file or directory”。
Sorry, I typed that wrong :)  Try `cat /etc/nv_tegra_release`

> `[network] getHostByName() unable to resolve host 'f309bb28-0124-43c3-ae1e-8695880a7948.local'`

If you are using Chrome, in `chrome://flags` can you try disabling `#enable-webrtc-hide-local-ips-with-mdns`

tchy, 2023-04-25:
> Sorry, I typed that wrong :) Try `cat /etc/nv_tegra_release`
>
>
> If you are using Chrome, in `chrome://flags` can you try disabling `#enable-webrtc-hide-local-ips-with-mdns`

`#enable-webrtc-hide-local-ips-with-mdns` is work.
`cat /etc/nv_tegra_release` show `# R35 (release), REVISION: 2.1, GCID: 32413640, BOARD: t186ref, EABI: aarch64, DATE: Tue Jan 24 23:38:33 UTC 2023`

dusty-nv, 2023-04-25:
> `#enable-webrtc-hide-local-ips-with-mdns` is work.

OK thanks, glad you got it working - on some local networks I see that mDNS (the `.local` domains) take longer to propagate or have trouble working.   Did you only encounter this behavior when running inside the container?

alec-g, 2023-09-23:
Hello,

I am having a similar issue, in my scenario, I can only connect to my Jetson device via a private VPN connection, disabling the `#enable-webrtc-hide-local-ips-with-mdns` flag in chrome does not work, is there a workaround for this for VPN connections?

alec-g, 2023-09-25:
Resolved by removing STUN server on local connections.

SeeRich, 2023-11-12:
@alec-g could you provide some more information on how you removed the stun server?
Thank you!

Hongfeng-mtec, 2023-12-13:
> Resolved by removing STUN server on VPN / local connections.

I am also experiencing issue with VPN, and would like to know how you "removed the STUN server". Thanks.

poett1, 2024-06-17:
@SeeRich @Hongfeng-mtec there's a [new commit regarding the stun STUN server ](https://github.com/dusty-nv/jetson-utils/commit/72c4e095f0575091992a5803d79af08e138e3369#diff-24bd0690ab02288d2458df98626a75be5041632777b3374ded31fa5df369c838). `video-viewer --stun-server=disabled /dev/video0 webrtc://@:8554/my_stream` should disable it. However, I'm still facing the same issue.

yshuolu, 2025-03-26:
VPN does not work, even if disable the chrome feature [#enable-webrtc-hide-local-ips-with-mdns](chrome://flags/#enable-webrtc-hide-local-ips-with-mdns)

enginger, 2025-07-02:
I am facing the same issue when it comes to loading the webui video with errors

ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Fatal Python error: Aborted

and

[gstreamer] gstreamer webrtcbin-0 ERROR Your GStreamer installation is missing a plug-in.
[gstreamer] gstreamer Debugging info: ../ext/webrtc/gstwebrtcbin.c(146): _have_nice_elements (): /GstPipeline:pipeline1/GstWebRTCBin:webrtcbin-0:
libnice elements are not available

could i get some help on this? thank you @yshuolu @dusty-nv @tchy @SeeRich @alec-g