dusty-nv/jetson-inference #1620: Not getting option to install models during using docker or manual build

hpetty3d, 2023-04-24:
So I've been trying to get up and running on my Jetson Nano Developer Kit and followed your youtube videos for installing the hello ai world project. Every video shows a part where you have the option to install the various models. However no matter what method I used I'm not seeing this screen. Did something change since these videos came out?

Thanks!

dusty-nv, 2023-04-24:
Hi @hpetty3d, sorry yes - this changed, and you no longer need to explicitly download models because the library will automatically download them for you on-demand the first time you try and run a model.   References to the model-downloader.sh tool have been removed from the docs, but yea they are still in those YouTube videos.

If you run imagenet/imagenet.py or detectnet/detectnet.py on a test image, does it download the model(s) for you okay?

hpetty3d, 2023-04-24:
Ah ok! Good to know I'm not crazy. I hadn't gotten that far because I was racking my brain over the model downloads LOL. Thanks for the quick reply. Next time I'm at my computer I'll test it out and let you know but I'm sure it works!

Thanks a bunch!

hpetty3d, 2023-04-24:
It appears to be working fine! Only issue I'm running into is that the output image doesn't stay open after the script finishes. It opens a window and then closes immediately so I just have to go find it in data/images/test/. Not sure if this is correct or if I may need to install something to get it to stay open.

dusty-nv, 2023-04-25:
> Not sure if this is correct or if I may need to install something to get it to stay open.

@hpetty3d yes that is expected, it closes the window showing the image because the program exits (this is more typically of camera streaming or doing batch processing).  You would have to add the equivalent of `sleep(10)` to the end of the program or wait for user input until it exited.

closed 2023-04-27 by dusty-nv

CamRoni1339, 2024-03-04:
> Hi @hpetty3d, sorry yes - this changed, and you no longer need to explicitly download models because the library will automatically download them for you on-demand the first time you try and run a [...]

@dusty-nv 

Hello! I am trying to use a different model for this program (specifically trafficcamnet) and I am running into a problem where it won't work. I tried running the piece of code that you talked about but it is not installing the models and instead is telling me that detectNet failed to initialize. If you could explain more about the process to install these models or give me another solutions for installing a different model it would be much appreciated.

Thanks!