dusty-nv/jetson-inference #1622: CUDA error after follow solution for [OpenGL] failed to create X11 Window.

Grissanapong27, 2023-04-26:
Hi, I'm try to run python example with jetson nano and MIPI csi camera but display not show from [OpenGL] failed to create X11 Window
[image]

 and it also show 
[image]
then i follow this solution #619 but after I done it this error happen
[image]
[image]
Thank you in advance for any help.

Grissanapong27, 2023-04-26:
oh I see, I didn't clone the dev banch but are there any solution to fix it ,
I'm also found sample code from https://forums.developer.nvidia.com/t/video-output-does-not-show-up-because-opengl-failed-to-create-x11-window-imagenet-camera-failed-to-create-opengl-display/68681/10
but i'm still face same error

dusty-nv, 2023-04-26:
Hi @Grissanapong27, does video-viewer/video-viewer.py or detectnet/detectnet.py work for you?

In your detectnet2.py script, can you try creating the videoOutput interface first, before the videoSource or detectNet objects?

Also, you have a physical display attached to your Jetson right?

Grissanapong27, 2023-04-26:
Thank you for your reply. The video-viewer and detectnet are working, but the detectnet.py and other .py script can detect objects but can't displaying the video output. Yes, I have a physical display connected to my Jetson Nano, I don't have my Jetson Nano with me right now, but I will make sure to try creating the videoOutput interface before the videoSource. Once I do so, I will update you on the results. Thanks .

Grissanapong27, 2023-04-27:
@dusty-nv after I try Video-viewer it work but when i run video-viewer.py csi://0 it pop-up this window 
[image]
then it shutdown
and here log
[video-viewer log.txt](https://github.com/dusty-nv/jetson-inference/files/11339062/video-viewer.log.txt)

dusty-nv, 2023-04-27:
``` 
[OpenGL] creating 1280x720 texture (GL_RGB8 format, 2764800 bytes)
[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/smr/jetson-inference/utils/display/glTexture.cpp:360
[cuda]   cudaGetLastError()
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/smr/jetson-inference/utils/cuda/cudaYUV-NV12.cu:154
[cuda]   cudaNV12ToRGB(input, (uchar3*)output, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/smr/jetson-inference/utils/cuda/cudaColorspace.cpp:42
[cuda]   cudaConvertColor(latestYUV, mFormatYUV, nextRGB, format, mOptions->width, mOptions->height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
```

I think there might be something awry with your graphics drivers or X11/display setup.   You aren't trying to view this remotely over VNC or SSH tunneling right?   Can you see if you have the same issue when running the [Docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md)?   If so, I would probably flash another SD card and try a fresh install.

Grissanapong27, 2023-04-28:
@dusty-nv I usually remote in from the VNC viewer, but even though I run the script from the Jetson Nano, it still has the same issue. However, when running the Docker container and mounting the example code and my code into the container, the code works fine.Now I'm work on project that work with ROS how can I mount my package in to container?

dusty-nv, 2023-04-28:
> However, when running the Docker container and mounting the example code and my code into the container, the code works fine.

OK - my guess is that at some point, some package got installed onto your Nano that conflicts with the native OpenGL/X11 graphics drivers.   You can try installing mesa-utils and running glxgears / glxinfo to see if that helps diagnose what is going on, and if you might need to reflash your SD card.

> Now I'm work on project that work with ROS how can I mount my package in to container?

What I would do is use one of the `ros:*-pytorch-*` containers from https://github.com/dusty-nv/jetson-containers . For example, `dustynv/ros:foxy-pytorch-l4t-r32.7.1`.   These `ros:*-pytorch-*` containers are based on the jetson-inference container, so they have jetson-inference included in them already.

Grissanapong27, 2023-05-01:
@dusty-nv Thanks for your advice,After I try to run glxgears can work and here glxinfo 
[glxinfo.txt](https://github.com/dusty-nv/jetson-inference/files/11363822/glxinfo.txt)