dusty-nv/jetson-inference #1623: Error with using TAO model

TomGrimwood, 2023-04-26:
Using an NVIDIA Jetson AGX Orin Developer Kit.
```bash
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ docker/run.sh
$ cd build/aarch64/bin
# Download test video
$ wget https://nvidia.box.com/shared/static/veuuimq6pwvd62p9fresqhrrmfqz0e2f.mp4 -O pedestrians.mp4

# Python
$ ./detectnet.py --network=peoplenet pedestrians.mp4 ./images/test/pedestrians_ssd.mp4
```

```html
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for pedestrians.mp4
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 960x540  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)960, height=(int)540, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=pedestrians.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///jetson-inference/build/aarch64/bin/pedestrians.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///jetson-inference/build/aarch64/bin/pedestrians.mp4
     - protocol:  file
     - location:  pedestrians.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      960
  -- height:     540
  -- frameRate:  29.97
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
failed to find/open file /proc/device-tree/model
[gstreamer] gstEncoder -- detected board 'Jetson AGX Orin'
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 maxperf-enable=1 ! video/x-h264 ! h264parse ! qtmux ! filesink location=./images/test/pedestrians_ssd.mp4 
[video]  created gstEncoder from file:///jetson-inference/build/aarch64/bin/./images/test/pedestrians_ssd.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///jetson-inference/build/aarch64/bin/./images/test/pedestrians_ssd.mp4
     - protocol:  file
     - location:  ./images/test/pedestrians_ssd.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      H264
  -- codecType:  v4l2
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  2560x1440
[OpenGL] glDisplay -- X window resolution:    2560x1440
[OpenGL] glDisplay -- display device initialized (2560x1440)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      2560
  -- height:     1440
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be string, but is null
Aborted (core dumped)
```

Using any of the TAO models has this error, but the inception and mobilenet network options do not. What has happened here?

dusty-nv, 2023-04-26:
Hi @TomGrimwood, it appears to be a JSON parsing error of the [`jetson-inference/data/networks/models.json`](https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/models.json) file, however I'm unable to reproduce it - can you try re-downloading that file in case it got corrupted?

What is the version of JetPack-L4T that you are running?  (you can check this with `cat /etc/nv_tegra_release`)    I wonder if your container needs updated...can you try [building from source](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) and seeing if you have this problem outside of container?

TomGrimwood, 2023-04-28:
Thanks Dustin,

I had an image from your jetson-containers downloaded, so I docker image prune -a  and got rid of everything and it now seems to be working a treat. Thanks.

I notice on https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md it doesnt list 35.3.1, but I see you have a docker image for it anyway.

LaurenMcGarry1, 2023-11-01:
Hi @dusty-nv ,

I am getting the exact same error, using the same kit.
I have checked for updates in the repo. 
I have tried the 'docker image prune -a'
then re ran the docker container, cd into build/aarch64/bin, and run 'detectnet.py --model=peoplenet pedestrians.mp4 pedestrians_peoplenet.mp4'

Is there anything else that I can try?

I am running R34, REVISION: 1.1

Thanks in advance.

dusty-nv, 2023-11-02:
@TomGrimwood try re-downloading this file:  https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/models.json

Also, you are running an old developer preview release of JetPack 5, you should probably update it to the latest.