dusty-nv/jetson-inference #1627: failed to create videoSource device when Run $ imagenet.py /dev/video1

mmhzlrj, 2023-04-28:
Hi @dusty-nv. I am learning the imagenet and trying to use it with my USB camera. But something went wrong.  
 
initialized.
[TRT]    tensorNet::LoadClassLabels() failed to find networks/Googlenet/networks/ilsvrc12_synset_words.txt
[TRT]    loaded 0 class labels
[TRT]    didn't load expected number of class descriptions  (0 of 1000)
[TRT]    filling in remaining 1000 class descriptions with default labels
[TRT]    imageNet -- networks/Googlenet/bvlc_googlenet.caffemodel initialized.
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video1

(python3:12063): GStreamer-CRITICAL **: 18:31:13.476: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed

(python3:12063): GStreamer-CRITICAL **: 18:31:13.476: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed
[gstreamer] gstCamera -- didn't discover any v4l2 devices
[gstreamer] gstCamera -- device discovery and auto-negotiation failed
[gstreamer] gstCamera -- failed to create device v4l2:///dev/video1
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 59, in <module>
    input = videoSource(args.input, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
jet@jet-nano:~/Desktop$ clear

jet@jet-nano:~/Desktop$ imagenet.py /dev/video1

imageNet -- loading classification network model from:
         -- prototxt     networks/Googlenet/googlenet.prototxt
         -- model        networks/Googlenet/bvlc_googlenet.caffemodel
         -- class_labels networks/Googlenet/networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   1

[TRT]    TensorRT version 8.2.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - caffe  (extension '.caffemodel')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +230, GPU +0, now: CPU 254, GPU 1933 (MiB)
[TRT]    [MemUsageSnapshot] Begin constructing builder kernel library: CPU 254 MiB, GPU 1935 MiB
[TRT]    [MemUsageSnapshot] End constructing builder kernel library: CPU 284 MiB, GPU 1942 MiB
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    found engine cache file /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel.1.1.8201.GPU.FP16.engine
[TRT]    found model checksum /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel.sha256sum
[TRT]    echo "$(cat /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel.sha256sum) /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel" | sha256sum --check --status
[TRT]    model matched checksum /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel.sha256sum
[TRT]    loading network plan from engine cache... /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel.1.1.8201.GPU.FP16.engine
[TRT]    device GPU, loaded /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 275, GPU 1935 (MiB)
[TRT]    Loaded engine size: 20 MiB
[TRT]    Using cublas as a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +158, GPU -8, now: CPU 433, GPU 1921 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +240, GPU -4, now: CPU 673, GPU 1917 (MiB)
[TRT]    Deserialization required 9140138 microseconds.
[TRT]    [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +19, now: CPU 0, GPU 19 (MiB)
[TRT]    Using cublas as a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU -5, now: CPU 673, GPU 1916 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +0, GPU +1, now: CPU 673, GPU 1917 (MiB)
[TRT]    Total per-runner device persistent memory is 14380032
[TRT]    Total per-runner host persistent memory is 89952
[TRT]    Allocated activation device memory of size 3612672
[TRT]    [MemUsageChange] TensorRT-managed allocation in IExecutionContext creation: CPU +0, GPU +18, now: CPU 0, GPU 37 (MiB)
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       68
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 3612672
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'data'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  3
                -- dim #0  3
                -- dim #1  224
                -- dim #2  224
[TRT]       binding 1
                -- index   1
                -- name    'prob'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  3
                -- dim #0  1000
                -- dim #1  1
                -- dim #2  1
[TRT]    
[TRT]    binding to input 0 data  binding index:  0
[TRT]    binding to input 0 data  dims (b=1 c=3 h=224 w=224) size=602112
[TRT]    binding to output 0 prob  binding index:  1
[TRT]    binding to output 0 prob  dims (b=1 c=1000 h=1 w=1) size=4000
[TRT]    
[TRT]    device GPU, /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel initialized.
[TRT]    tensorNet::LoadClassLabels() failed to find networks/Googlenet/networks/ilsvrc12_synset_words.txt
[TRT]    loaded 0 class labels
[TRT]    didn't load expected number of class descriptions  (0 of 1000)
[TRT]    filling in remaining 1000 class descriptions with default labels
[TRT]    imageNet -- networks/Googlenet/bvlc_googlenet.caffemodel initialized.
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video1

(python3:12480): GStreamer-CRITICAL **: 18:36:09.110: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed

(python3:12480): GStreamer-CRITICAL **: 18:36:09.113: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed
[gstreamer] gstCamera -- didn't discover any v4l2 devices
[gstreamer] gstCamera -- device discovery and auto-negotiation failed
[gstreamer] gstCamera -- failed to create device v4l2:///dev/video1
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 59, in <module>
    input = videoSource(args.input, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2023-04-28:
> [gstreamer] gstCamera -- didn't discover any v4l2 devices
> [gstreamer] gstCamera -- device discovery and auto-negotiation failed
> [gstreamer] gstCamera -- failed to create device v4l2:///dev/video1

@mmhzlrj can you do `ls /dev/video*` and list your V4L2 devices?

Also, can you try running this to confirm that your V4L2 camera is reporting it's video formats okay?

``` bash
$ sudo apt-get install v4l-utils
$ v4l2-ctl --device=/dev/video0 --list-formats-ext
$ v4l2-ctl --device=/dev/video1 --list-formats-ext
```

mmhzlrj, 2023-04-29:
> > [gstreamer] gstCamera -- didn't discover any v4l2 devices
> > [gstreamer] gstCamera -- device discovery and auto-negotiation failed
> > [gstreamer] gstCamera -- failed to create device v4l2:///dev/video1
> 
> @mmhzlrj can you do `ls /dev/video*` and list your V4L2 devices?
> 
> Also, can you try running this to confirm that your V4L2 camera is reporting it's video formats okay?
> 
> ```shell
> $ sudo apt-get install v4l-utils
> $ v4l2-ctl --device=/dev/video0 --list-formats-ext
> $ v4l2-ctl --device=/dev/video1 --list-formats-ext
> ```
jet@jet-nano:~$ ls /dev/video*
/dev/video0  /dev/video1

jet@jet-nano:~$ sudo apt-get install v4l-utils
[sudo] password for jet: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l-utils is already the newest version (1.14.2-1).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dpkg-repack efibootmgr gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common
  kde-window-manager kinit kio kpackagetool5 kwayland-data kwin-common
  kwin-data kwin-x11 libdebian-installer4 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5designer5
  libqt5help5 libqt5multimedia5 libqt5multimedia5-plugins
  libqt5multimediaquick-p5 libqt5multimediawidgets5 libqt5opengl5
  libqt5quickwidgets5 libxcb-composite0 libxcb-cursor0 libxcb-damage0
  os-prober python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
jet@jet-nano:~$ v4l2-ctl --device=/dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB
		Size: Discrete 3264x2464
			Interval: Discrete 0.048s (21.000 fps)
		Size: Discrete 3264x1848
			Interval: Discrete 0.036s (28.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1640x1232
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)

jet@jet-nano:~$ v4l2-ctl --device=/dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
[... 329 lines cut ...]
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1792x1008
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)

Anything wrong?@dusty-nv

mmhzlrj, 2023-04-29:
Hi, @dusty-nv . I think I got a new issue by using imagenet to detect fruits images. I got `class #950` on top left side of the images. Does it suppose to be the name of a fruit?

[image]
I was runing `$ imagenet fruit output` 
The `fruit` folder as the URI of the input stream which has fruits images from /jetson-inference/data/images.
And the `output` folder as the URI of the output stream.

dusty-nv, 2023-05-01:
> ` imagenet fruit --network=resnte-18`
@mmhzlrj try running it with `--network=resnet-18` instead

> I got `class #950` on top left side of the images. Does it suppose to be the name of a fruit?

I'm not sure why it's not loading the class names.  These get loaded from [`jetson-inference/data/networks/ilsvrc12_synset_words.txt`](https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/ilsvrc12_synset_words.txt)

If it continues to be an issue, can you post the console log from running imagenet/imagenet.py here?

mmhzlrj, 2023-05-02:
Hi, @dusty-nv 

[log.txt](https://github.com/dusty-nv/jetson-inference/files/11369624/log.txt)

[image]

> @mmhzlrj try running it with `--network=resnet-18` instead
>
>
> I'm not sure why it's not loading the class names. These get loaded from [...]

dusty-nv, 2023-05-02:
> `~/jetson-inference/data/images$  imagenet fruit --network=resnet-18 output`

Hi @mmhzlrj, I just ran this exact command from your log, and it loaded/used the class labels without issue.   If it keeps happening, you might want to make sure you are using the latest code by running a `git pull --recurse-submodules` and recompiling/reinstalling jetson-inference.

> I think I am missing this file inference/data/networks/ilsvrc12_synset_words.txt
> Where can I download this file?

Do you have it at  `~/jetson-inference/data/networks/ilsvrc12_synset_words.txt`?  If not, you can download it from https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/ilsvrc12_synset_words.txt

mmhzlrj, 2023-05-03:
> 
Hi,@dusty-nv. I don't have it at ~/jetson-inference/data/networks/ilsvrc12_synset_words.txt. The result ran great after I downloaded file and put it to the right path. Thank you so much.
[image]

closed 2023-05-03 by mmhzlrj