dusty-nv/jetson-inference #1628: How to map depth values to original image coordinates

DigeSegun, 2023-04-28:
How do I map the depth values to the original image coordinates in python for depthnet.py

dusty-nv, 2023-04-28:
@DigeSegun do you mean how to get the Z-component given the camera calibration, or how to resize the raw depth field to the original image size?    If the later, you can simply scale the pixel coordinates linearly or resize the depth field image.

dige-mothership, 2023-04-28:
@dusty-nv Thanks for responding to this. Our end goal is to filter out parts of the image that are further away from the camera. Do you think the most efficient way is to resize the resultant image (with the overlayed depthfield) and then extract the closest coordinates of the image based on the colors?

dusty-nv, 2023-04-28:
@dige-mothership I think the most efficient way would be to just re-scale the pixel coordinates that you use to lookup the raw depth field.   Like the scaling factors would be:

``` python
scale_x = float(net.GetDepthFieldWidth()) / float(img.width)
scale_y = float(net.GetDepthFieldHeight()) / float(img.height)
```

Then when you are indexing the depth field values, simply multiply these scalars by the image pixel coordinates and cast them to int()

dige-mothership, 2023-04-28:
Thanks @dusty-nv !

closed 2023-04-28 by DigeSegun

dige-mothership, 2023-06-28:
@dusty-nv  This was my implementation of the above but i think im doing something wrong, could you please take a look?

`import sys
import argparse
import jetson.inference
import jetson.utils
from jetson_inference import depthNet
from jetson_utils import videoSource, videoOutput, cudaOverlay, cudaDeviceSynchronize, Log
import numpy as np
from depthnet_utils import depthBuffers
import cv2

# parse the command line
parser = argparse.ArgumentParser(description="Mono depth estimation on a video/image stream using depthNet DNN.", 
                                 formatter_class=argparse.RawTextHelpFormatter, 
                                 epilog=depthNet.Usage() + videoSource.Usage() + videoOutput.Usage() + Log.Usage())

parser.add_argument("input", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="fcn-mobilenet", help="pre-trained model to load, see below for options")
parser.add_argument("--visualize", type=str, default="input,depth", help="visualization options (can be 'input' 'depth' 'input,depth'")
parser.add_argument("--depth-size", type=float, default=1.0, help="scales the size of the depth map visualization, as a percentage of the input size (default is 1.0)")
parser.add_argument("--filter-mode", type=str, default="linear", choices=["point", "linear"], help="filtering mode used during visualization, options are:\n  'point' or 'linear' (default: 'linear')")
parser.add_argument("--colormap", type=str, default="inferno", help="colormap to use for visualization (default is 'viridis-inverted')",
                                  choices=["inferno", "inferno-inverted", "magma", "magma-inverted", "parula", "parula-inverted", 
                                           "plasma", "plasma-inverted", "turbo", "turbo-inverted", "viridis", "viridis-inverted"])

try:
	args = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the segmentation network
net = depthNet(args.network, sys.argv)
depth_field = net.GetDepthField()
# create buffer manager
buffers = depthBuffers(args)

# create video sources & outputs
input = videoSource(args.input, argv=sys.argv)
output = videoOutput(args.output, argv=sys.argv)
depth_numpy = jetson.utils.cudaToNumpy(depth_field)
# process frames until EOS or the user exits
while True:
    # capture the next image
    img_input = input.Capture()
    img = cv2.imread(args.input)

    if img_input is None: # timeout
        continue
        
    # allocate buffers for this size image
    buffers.Alloc(img_input.shape, img_input.format)

    # process the mono depth and visualize
    net.Process(img_input, buffers.depth, args.colormap, args.filter_mode)

    # composite the images
    if buffers.use_input:
        cudaOverlay(img_input, buffers.composite, 0, 0)
        
    if buffers.use_depth:
        cudaOverlay(buffers.depth, buffers.composite, img_input.width if buffers.use_input else 0, 0)

    # render the output image
    output.Render(buffers.composite)

    # update the title bar
    # output.SetStatus("{:s} | {:s} | Network {:.0f} FPS".format(args.network, net.GetNetworkName(), net.GetNetworkFPS()))

    # print out performance info
    cudaDeviceSynchronize()
    net.PrintProfilerTimes()

    #lets say we are trying to get the depth values for X - 500 and y 640, we would do it like this
    scale_x = float(net.GetDepthFieldWidth()) / float(img_input.width)
    scale_y = float(net.GetDepthFieldHeight()) / float(img_input.height)
    print(img_input.width)
    depth_list = []
    height, width, _ = img_input.shape
    
    # print(type(img_input.shape))
    for x in range(width):
        for y in range(height):
            new_scale_x = int(scale_x * x)
            new_scale_y = int(scale_y * y)
            depth_list.append(depth_field[new_scale_x ,new_scale_y])

    percent = max(depth_list) * 0.4
    for x in range(width):
        for y in range(height):
            new_scale_x = int(scale_x * x)
            new_scale_y = int(scale_y * y)
            if depth_numpy[new_scale_x ,new_scale_y] < percent: 
                cv2.circle(img, (x, y), 50, (0, 255, 0), thickness=1)

    if not input.IsStreaming() or not output.IsStreaming():
        break
cv2.imwrite('output/depth_testing.jpg', img)`

dusty-nv, 2023-06-29:
Hi @dige-mothership, I can't speak to all the changes, but I think it should be `depth_field[new_scale_y, new_scale_x]` as per https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#accessing-image-data-in-python

(I think the same for the `depth_numpy` indexing too - the subscript array indices are ordered by dimension)

dige-mothership, 2023-06-29:
@dusty-nv thank you! do you have any example of vectorized numpy operations in jetson_inference?

dusty-nv, 2023-06-29:
The segnet_utils.py do stuff on a mapped cudaImage with numpy operations (to calculate the histogram)

By "vectorized numpy operations" I basically just mean the numpy functions like numpy.add, array slicing, ect which are implemented with C/C++ underneath