dusty-nv/jetson-inference #1633: Hello-  Models not included

ahmet-f-gumustas, 2023-05-03:
Hi ;

I 'm using jetson nano and jetson jetpack R32.7.3 but ı don't understand why , source installation not working.
- I'm doing cmake ../ but I don't get the required output to download the models why, I checked everything necessary.
- just " Do you want to install pytorch only? "

dusty-nv, 2023-05-03:
Hi @ahmet-f-gumustas, the Model Downloader tool is no longer necessary to use, because the models will now automatically be downloaded on-demand when you load them with imagenet/detectnet/segnet/ect.   All of the info used to download them is now stored in [`jetson-inference/data/networks/models.json`](https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/models.json)

If you just run imagenet/detectnet/segnet/ect does it download the model(s) okay for you?

ahmet-f-gumustas, 2023-05-04:
Thank you so much i will check and return ..

ahmet-f-gumustas, 2023-05-04:
I watched the videos below and am trying to learn to understand the application. but I couldn't load the models like here. actually i'm installing as source so i don't use docker

- https://www.youtube.com/watch?v=5rbOsKCZ-VU
- https://www.youtube.com/watch?v=mB025B7KpeE

I mean, when I do the steps in the videos, I don't get the same output even when I first import it. strange situation  :)

dusty-nv, 2023-05-04:
@ahmet-f-gumustas what is the code you are trying to run?

Can you try running imagenet.py/detetnet.py like from https://github.com/dusty-nv/jetson-inference#inference and see if those work first?

ahmet-f-gumustas, 2023-05-09:
@dusty-nv  I'am sory dusty :)

Sorry for taking your time. It worked after a while. thanks for your help .

closed 2023-05-09 by ahmet-f-gumustas