dusty-nv/jetson-inference #1637: How to get  URI of current frame

DigeSegun, 2023-05-08:
Hello I was wondering for the frames that are being looped over for example in posenet if I pass a folder of images. How do I access the current URI of the img I'm currently on

dusty-nv, 2023-05-08:
@DigeSegun unfortunately this isn't something that's exposed through the videoSource interface, sorry about that - it was primarily intended for video/camera streams which don't have unique frame URI's.  In theory you could add a `const char* GetFrameURI()` function to videoSource that [imageLoader](https://github.com/dusty-nv/jetson-utils/blob/0bcb19b498326eb866a80d7d13388b2e59bc9dfd/image/imageLoader.cpp) sets.

dige-mothership, 2023-08-08:
@DigeSegun is this good to close?