dusty-nv/jetson-inference #1638: CSI CAMERA ISSUE UNDER YOLOV7

akhandvaibhav17, 2023-05-11:
My team was trying to run YOLOv7 detect.py file on jetson nano. After giving the command "python3 detect.py source -0 weights yolov7.pt" the CSI Camera launches with warning "unable to get camera fps" and shows green screen. The is no issue with the camera as "nvgstcapture" command is working fine. Please help finding the solution to the issue. 
[image]
[image]
[image]
[image]

dusty-nv, 2023-05-11:
@akhandvaibhav17 it looks like you are using OpenCV video capture and YOLOv7 detect.py code is not from this repo, I would suggest that you follow up on the Jetson Nano forums:  https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76

At a glance, you are having OpenCV capture video from `/dev/video0` (V4L2), whereas for MIPI CSI camera you should provide cv2.VideoCapture() with a GStreamer pipeline that uses nvarguscamerasrc element:   https://github.com/JetsonHacksNano/CSI-Camera/blob/master/simple_camera.py

closed 2023-05-11 by dusty-nv

gaumh, 2024-03-06:
Hello, How did you solve this problem?
Because I have the same problem, I would like to trouble you to answer it again, Thank you!

akhandvaibhav17, 2024-03-06:
Well, this problem wasn't solved though any of the solutions mentioned on internet. Instead we switched to a USB camera in place of a CSI camera and it worked properly. We couldn't find out the issue with CSI camera but it might be that the camera is defective or it can't handle the required fps.