dusty-nv/jetson-inference #1639: v4l2 videoSource is not loading, GStreamer getting paused.

nithinpremkumar1999, 2023-05-12:
Hi, I use a Logitech C270 webcam, I have built from source this project on a Jetson Nano with jetpack 4.6.1.
My code:

from jetson_utils import videoSource, videoOutput

camera = videoSource("v4l2:///dev/video0")
display = videoOutput("display://0") 

while display.IsStreaming():
    img = camera.Capture()

    display.Render(img)
    display.SetStatus("Object Detection | Network {:.0f} FPS".format(self.net.GetNetworkFPS()))

The error: 
[image]

The camera works fine on cheese, this problem doesn't happen when I run py files from the project, just py files written by me are failing to start the camera.  Any help will be greatly appreciated. Because of this I cant use detectNet.detect(img) because it says it expects cudaimage.

dusty-nv, 2023-05-12:
@nithinpremkumar1999 try changing your code to this:

``` python
from jetson_utils import videoSource, videoOutput

camera = videoSource("v4l2:///dev/video0")
display = videoOutput("display://0") 

while display.IsStreaming():
    img = camera.Capture()

    if img is None:
        continue

    display.Render(img)
    display.SetStatus("Object Detection | Network {:.0f} FPS".format(self.net.GetNetworkFPS()))
```

There was a change where videoSource.Capture() will return `None` in the event of a timeout, so try adding that `if img is None` check to your code.

nithinpremkumar1999, 2023-05-12:
Thank you for your speedy response, I will try the code (presently I am away from Jetson). But I do want to know why there is a timeout. Is it because of some webcam codec issue? I had read somewhere someone else have that issue.
Here is a screenshot about details of my webcam
[image]
[image]

dusty-nv, 2023-05-12:
> But I do want to know why there is a timeout.

sometimes when the camera is first starting it's normal to have a timeout, especially when the video is compressed or it's a network stream

nithinpremkumar1999, 2023-05-12:
Okay thank you for the insight I will get back to you if the solution worked or not :)

nithinpremkumar1999, 2023-05-15:
@dusty-nv Thank you for the help! Your solution worked :)