dusty-nv/jetson-inference #1640: AttributeError: 'Series' object has no attribute 'iteritems'

githubwalt, 2023-05-13:
Hi Dusty,

First of all, great job!

I'm new to all of this.
While following your video on Training Object Detection Models, I had this error while Downloading the Data.

File "open_images_downloader.py", line 212, in <module>
    log_counts(annotations[dataset_type]['ClassName'])
  File "open_images_downloader.py", line 80, in log_counts
    for k, count in values.value_counts().iteritems():
  File "/usr/local/lib/python3.8/dist-packages/pandas/core/generic.py", line 5989, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'Series' object has no attribute 'iteritems'

I found a solution by changing line 80
"count in values.value_counts().iteritems():"
to
"count in values.value_counts().items():"

This took me a while though because when I opened "jetson-inference/python/training/detection/ssd/open_images_downloader.py" via the file browser, any changes I made there had no effect!?
Only "nano open_images_downloader.py" via the terminal did the job.

So my questions are:
1. could you please update the file for future users.  Or is there another reason for the error?
2. why couldn't I edit the file in the text editor via the file browser?  (it seamed to save the file without any warning)

Best regards,
Walter

githubwalt, 2023-05-13:
Now that I'm here anyway, I'm a little confused so one more little question:
My jetson is busy 'Re-training SSD-Mobilenet', and after some more hours I will have my model "ssd-mobilenet.onnx".
I previously followed this: 'Coding Your Own Object Detection Program' and have my "my-detection.py" which I could run on my USB camera.

Now, to run "my-detection.py" on that new model "ssd-mobilenet.onnx",
in that file, do I simply have to replace :
"net = jetson.inference.detectNet("**ssd-mobilenet-v2**", threshold=0.5)"
by
"net = jetson.inference.detectNet("**ssd-mobilenet.onnx**", threshold=0.5)"
?
Or how do I proceed?

githubwalt, 2023-05-14:
Now that I'm here anyway, I'm a little confused so one more little question:
I retrained following this 'Re-training SSD-Mobilenet', and have my model "ssd-mobilenet.onnx".
I previously followed this: 'Coding Your Own Object Detection Program' and have my "my-detection.py" which I could run on my USB camera.  I did some changes there to scale the output etc.

How do I proceed to run "my-detection.py" on that new model "ssd-mobilenet.onnx"?

dusty-nv, 2023-05-15:
> I read somewhere that the newer version of pandas doesn't have iteritems() any longer but now uses items() instead.

Thanks @githubwalt, it appears that `iteritems()` had been deprecated and removed in Pandas 2.0, but `items()` was available as far back as Pandas 1.0, so I just changed it to `items()` in https://github.com/dusty-nv/pytorch-ssd/commit/57c901e3e3abf33c64dffc9292e7eb8458397ac6

closed 2023-05-15 by dusty-nv

githubwalt, 2023-05-16:
Thanks,
As I said, I changed line 80, but the problem is that this change is not permanent.   Each time I re-enter the docker, the change I made is gone!?  Besides downloading the docker again, what is the right way or the steps to edit this file so that the change is permanent?

dusty-nv, 2023-05-16:
@githubwalt you could do a `docker commit` (from outside the container) and save your modifications to a new container tag, and then modify `docker/run.sh` to run that new tag instead.   Or you could make the code change outside of container to your local jetson-inference source tree (or pull my latest source) and mount that into the container and run that instead.

I don't know what version of L4T you are running, but I did update a bunch of the more recent jetson-inference container images with this fix, and think it would only re-download what changed if you were to `docker pull` it.

githubwalt, 2023-05-16:
@dusty-nv 
Thanks Dustin, I'll do that.  Like I said, I'm a novice so I constantly sit with a bunch of questions.
But I learn something every day 🙂

One more stupid question:
My project is to detect other aircraft while flying so for re-training I downloaded the Open Images dataset, classes Aircraft. 
But I can see that there are a lot of bad images/annotations in that set.  
By browsing the images, can I just delete those from the set?
I think not because there is probably more to that than just deleting images.
Or should I use a tool like Voxel51 FiftyOne?
Or is there any other way?

githubwalt, 2023-05-16:
Thank you for the swift reply Dustin, that's valuable information!

How about my first question:
.....
I have jetpack 5.1.1 and so L4T R35.3.1.
So can I just type "docker pull" or do I have to add the tag by typing "docker pull dustynv/jetson-inference:r35.3.1"?
This will not undo all the work I have done so far?

githubwalt, 2023-05-16:
Thanks again Dustin, worked like a charm!

githubwalt, 2023-05-30:
Hi Dustin,
I'm gathering images (airplanes) and would like to use V7's Auto-annotation tool to draw Polygons (Polygons seem to be the most accurate for object detection).
The problem is that the V7 export format of polygons is Darwin JSON, see: [https://docs.v7labs.com/docs/about-computer-vision-export-formats](url)
and your train_ssd.py does not support that format.
Would it be possible (hard) to include this format in train_ssd.py?
This would be great because I'm now finally getting familiar with your jetson-inference workflow here.
Please tell me if I'm wrong on assuming that polygons are better than simply drawing bounding boxes.

Walter

dusty-nv, 2023-05-30:
@githubwalt your typical object detection network like SSD, YOLO, ect supports bounding boxes (not polygons) and polygons would require some special kind of DNN (or may be a better fit for semantic segmentation).  Other types of models detect 3D bounding box (6-DoF).  These various kinds of outputs are inherent to the design of the network layers and topology.   You can probably find something in the research or code for PyTorch that supports what you want, but polygons aren't in jetson-inference.

If you just wanted to support bounding boxes in Darwin JSON, you should be able to modify train_ssd.py for it by making another dataloader like it has for Pascal VOC and OpenImages.

githubwalt, 2023-05-30:
Oh ok, of course,  I see.
I guess that I will have to stick to using cvat then.

githubwalt, 2023-06-02:
Hi Dustin,

I installed CVAT locally on a windows pc and annotated some 50 images, just to test the workflow.
I exported the dataset as PASCAL VOC 1.1.
In attachment is what CVAT generated.

Now I want to move that dataset to my Jetson to train a model, the same way as you describe on your page "Re-training SSD-Mobilenet".  (I did that successfully before with the open images dataset.)

The structure and arrangement of those generated PASCAL VOC files and folders are puzzling me.
If I look in train_ssd.py, to know which arguments I should use, things don't seem right to me.

So a few questions:
- Do I have to change anything in that dataset folder, like splitting up in train, validate and test ... or rename anything?
- Where does everything go on my Jetson?
- Do I have to change anything in train_ssd.py?
- What command should I now use to train my model?

Also this line on your page "Re-training SSD-Mobilenet"  confuses me:
"The base model was already pre-trained on a different dataset (PASCAL VOC) so that we don't need to train SSD-Mobilenet from scratch, which would take much longer. Instead we'll use transfer learning to fine-tune it to detect new object classes of our choosing."

Since I'm now going to use PASCAL VOC, will this change the above statement?

Thanks for your patience with me,
Walter

[aircraft_test_50.zip](https://github.com/dusty-nv/jetson-inference/files/11638224/aircraft_test_50.zip)

dusty-nv, 2023-06-02:
Hi Walter,

I don't believe you should have to change anything about your dataset.  You should be able to run train_ssd.py as shown here:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md#training-your-model

Having your dataset in Pascal VOC format is different than training the model on Pasal VOC itself - the data is different.

Good luck!

githubwalt, 2023-06-02:
Thank you Dustin.

I am confused because I read this article about training a model where they state that one has to split de dataset in to train and validate.
Here is the page:  [https://www.v7labs.com/blog/train-validation-test-set](https://www.v7labs.com/blog/train-validation-test-set)

In the PASCAL VOC dataset there isn't a train and validate split.
So, does the training on my dataset work in an entirely different way?
Could you please try to explain to me, being a novice?

dusty-nv, 2023-06-03:
CVAT makes a `default.txt` under ImageSets/Main that has all the images, which tends to be fine if you have a small number of images already and are just starting off training your model.  But yes, best practice would be to split that into train/val/test sets.

githubwalt, 2023-06-03:
Thanks.
But then I come back to my[ original question](https://github.com/dusty-nv/jetson-inference/issues/1640#issuecomment-1574204675) that the file/folder structure is different.

I plan to have a few thousand images to let it train on.
Just trying to understand the training process, and I want to do it right.

I dug a little into train_ssd.py and it looks like the splitting is already done in the code while training? (from line 280)
See: [https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/train_ssd.py#LL280C12-L280C12](https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/train_ssd.py#LL280C12-L280C12)

So is a validation images folder like in Open Images dataset not really necessary?
In other words, the validation images are not used while training? (I'm referring to "Re-training SSD-Mobilenet")

Am I wrong?

dusty-nv, 2023-06-04:
> I dug a little into train_ssd.py and it looks like the splitting is already done in the code while training? (from line 280)
> See: https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/train_ssd.py#LL280C12-L280C12

That is for if you have multiple independent datasets, it will combine all the train sets together.   Like if you had multiple datasets in Pascal VOC format.  I personally haven't used that feature, but could be useful if you were to capture multiple datasets with different things (or different iterations of the same dataset) and wanted to train your model across all of them.  It doesn't split it into train/test/val though.

> But then I come back to my[ original question](https://github.com/dusty-nv/jetson-inference/issues/1640#issuecomment-1574204675) that the file/folder structure is different.

The folder structure is the same, the only difference really is that under `ImageSets/Main` you need to create the `test.txt  train.txt  trainval.txt  val.txt` files from the master default.txt that CVAT outputs.  I would just make a Python script that randomly did this for you.

githubwalt, 2023-06-04:
Ok, I see.
I understand splitting the default.txt into test, train and val.txt,  **but what is trainval.txt used for**?
And if I understand correctly, the structure should become:
ImageSets
    - Main
         - train.txt(60% of default.txt)
         - val.txt (20% of default.txt)
         - test.txt (20% of default.txt)
         
So **without** the default.txt?

Thanks for being so patient with me and all my questions Dustin.
But I'm nearly there 😉
I really appreciate it!

githubwalt, 2023-06-05:
Is trainval.txt needed?
If yes, what is the purpose of it?

dusty-nv, 2023-06-05:
trainval.txt does indeed get used by default:   https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/vision/datasets/voc_dataset.py#L36

It uses the combined trainval.txt for training, and test.txt for test/validation.   You can change this in the code if you want something different.

Also, it will only fallback to using the default.txt from CVAT if it can't find trainval.txt/test.txt, so you needn't remove default.txt from your ImageSets/Main directory if you don't want to.

githubwalt, 2023-06-05:
Ok.
Looking in that voc_dataset.py file, I can't see any reference to 'train.txt' or 'val.txt', so they are not needed after all?

Also on line [50](https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/vision/datasets/voc_dataset.py#LL51C64-L51C64) the label_file_name  is set to 'labels.txt' which is called 'labelmap.txt' in my set.  So I DO have to change some things after all.
And if I change nothing at all, simply the 'default.txt' is used.

Sorry to keep on going on this, but I get mixed messages.  Very confusing.

Anyway, giving me that link to that file does help.  I will have to do it by "trial and error" I guess 😐

githubwalt, 2023-06-06:
Thanks Dustin.
I will try to follow your tutorial "Collecting your own Detection Datasets" and see what it produces.
Then I have my example structure.

dusty-nv, 2023-06-06:
@githubwalt you can also just download the original Pascal VOC dataset and inspect it's directory/file structure for reference:   http://host.robots.ox.ac.uk/pascal/VOC/voc2012/index.html#devkit

githubwalt, 2023-06-08:
Hi Dustin,
Just wanted to let you know that after following your tips I successfully trained a small set just for testing.

Meanwhile I gathered some 4000 images to do the real work.
I read somewhere that it's recommended to do some augmentation on the images.
I'm thinking of using the Albumentations git code for that.
But is this really necessary because looking here in [train_ssd.py](https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/train_ssd.py#L250) it seems that it is already doing some augmentation?

Is that correct?
What is your advise on this?

dusty-nv, 2023-06-08:
Hi @githubwalt, glad to hear that you got the workflow working!

Yes, train_ssd.py already does image augmentation:   https://github.com/dusty-nv/pytorch-ssd/blob/57c901e3e3abf33c64dffc9292e7eb8458397ac6/vision/ssd/data_preprocessing.py#L13

I haven't experimented with adding more (in my experience it hasn't been necessary), but that's up to you.  I guess I would probably train your model first and see how it goes.   You may also be interested in the `--validate-mean-ap` flag which prints out the per-class accuracies during the val step after each training epoch (this does take longer to run though)

githubwalt, 2023-06-08:
Thanks, glad I haven't got to bother with augmentation 🙂

(it's '--validation-mean-ap' 😉)

githubwalt, 2023-07-17:
Hi Dusty,

I'm about to train.

Question about argument '--balance-data', action='store_true'
It says 'Balance training data by down-sampling more frequent labels.'

I have 10 classes and one of them has the most images, maybe 80%, so hence has the more frequent labels.
This class is also the most important one.

So does this argument refer to labels in a particular image or does it refer to my classes.
If it refers to labels in all images of a particular class, I don't want to limit the training data of my most important class.

I hope my explanation makes sense to you.
What should I do and what would be the opposite of  action='store_true'?

Best regards,
Walter

githubwalt, 2023-07-17:
Thanks Dusty, but anyway, how would I disable that option?
action='store_false' maybe?

(I can't find the file where this option is used)

githubwalt, 2023-07-17:
Thanks Dusty.

githubwalt, 2023-07-18:
Another simple question @dusty-nv :

I'm training my model on resolution 1024 px because most of my images have 1 side of 1024 px and I want to detect objects from far away.  (You gave that advise, resolution of 512, in this thread: [https://github.com/dusty-nv/jetson-inference/issues/1660](url))
I also have a bunch of smaller images with the smallest being 128 px.
What happens to them, and other smaller images?
Do they get filled with a black background of 1024 x 1024 px?
Just making sure.

Thanks,
Walter

dusty-nv, 2023-07-18:
@githubwalt I believe all images will be resized to 1024x1024:

https://github.com/dusty-nv/pytorch-ssd/blob/6accaa88845ec135a7d6fe25e9a26afd4698639d/vision/transforms/transforms.py#L107

githubwalt, 2023-07-18:
And what if an image is not square?  Will it be stretched in the small direction?
I may be mixing things up, but I remember reading somewhere that a not square image gets filled instead of stretched to make it square.
I'm just trying to get my head around it.

Like I said earlier, the project is to detect other aircraft from a distance.
I have a lot of images of size 1024 with airplanes just barely visible, but visible.
So let's say that I don't change the training resolution so that it defaults to 300 x 300.
Will that airplane still be visible?
My reasoning is that the smaller the image, the smaller the objects, which is actually what I want.

Your answer in this thread confuses me:  [https://github.com/dusty-nv/jetson-inference/issues/1660](https://github.com/dusty-nv/jetson-inference/issues/1660)

I'm just wondering, in your opinion,  what would be the best training resolution for my project?

dusty-nv, 2023-07-18:
@githubwalt that Resize() code is using cv2 and I believe it doesn't crop/ect and just resizes to 1024x1024 (or whatever your training resolution is)

I haven't used above 512x512 resolution with SSD-Mobilenet so I'm not sure - regardless, it takes experimentation.   You might also find that the runtime inferencing performance of 1024x1024 is too slow for your requirements.