dusty-nv/jetson-inference #1645: "$ dpkg -l deepstream-5.0"  no packages found matching deepstream-5.0

mmhzlrj, 2023-05-16:
Hi @dusty-nv, I am learing how to use deepstream. Someone said if my jetpack version is over 4.5 then the jetpack 5.0 or above should be installed. But when I ran `dpkg -l deepstream-5.0` on my terminal it showed `no packages found matching deepstream-5.0`. How do I know if deepstream sdk has been installed or not for sure?  
 
$ sudo apt-cache show nvidia-jetpack
[sudo] password for jet: 
Package: nvidia-jetpack
Version: 4.6.3-b17
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-l4t-jetson-multimedia-api (>> 32.7-0), nvidia-l4t-jetson-multimedia-api (<< 32.8-0), nvidia-cuda (= 4.6.3-b17), nvidia-tensorrt (= 4.6.3-b17), nvidia-nsight-sys (= 4.6.3-b17), nvidia-cudnn8 (= 4.6.3-b17), nvidia-opencv (= 4.6.3-b17), nvidia-container (= 4.6.3-b17), nvidia-vpi (= 4.6.3-b17)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6.3-b17_arm64.deb
Size: 29368
SHA256: 694254a8667ebbf13852548bdd13a5b8ae61481ac059845b706398eefdcb9e01
SHA1: 67140fc8463ec61fd69352b225244b639c799edd
MD5sum: afa1382b6caded6b736d494fc481bab4
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

Package: nvidia-jetpack
Version: 4.6.2-b5
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-cuda (= 4.6.2-b5), nvidia-opencv (= 4.6.2-b5), nvidia-cudnn8 (= 4.6.2-b5), nvidia-tensorrt (= 4.6.2-b5), nvidia-visionworks (= 4.6.2-b5), nvidia-container (= 4.6.2-b5), nvidia-vpi (= 4.6.2-b5), nvidia-l4t-jetson-multimedia-api (>> 32.7-0), nvidia-l4t-jetson-multimedia-api (<< 32.8-0)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6.2-b5_arm64.deb
Size: 29378
SHA256: 925f4abff97e6024d86cff3b9e132e7c7554d05fb83590487381b7e925d5b2bb
SHA1: e3ef727e87df5c331aece34508c110d57d744fe9
MD5sum: 7cb2e387af41bc8143ac7b6525af7794
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

Package: nvidia-jetpack
Version: 4.6.1-b110
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-cuda (= 4.6.1-b110), nvidia-opencv (= 4.6.1-b110), nvidia-cudnn8 (= 4.6.1-b110), nvidia-tensorrt (= 4.6.1-b110), nvidia-visionworks (= 4.6.1-b110), nvidia-container (= 4.6.1-b110), nvidia-vpi (= 4.6.1-b110), nvidia-l4t-jetson-multimedia-api (>> 32.7-0), nvidia-l4t-jetson-multimedia-api (<< 32.8-0)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6.1-b110_arm64.deb
Size: 29366
SHA256: acfd9e75af780eab165361d61de4b4fe4974890864fe191060b402ac4c2f54d5
SHA1: a016568ac53705acc145a9f7e60505707bea259f
MD5sum: 79be976b184a8c885bd9169ea5b7fb7b
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

dusty-nv, 2023-05-16:
@mmhzlrj DeepStream isn't installed by default - if you do `apt-cache search deepstream`, can you find it?

``` bash
$ apt-cache search deepstream
deepstream-6.0 - Nvidia DeepStreamSDK runtime libraries, development files and samples
```

(this was on JetPack 4.6.1 / L4T R32.7.1)

mmhzlrj, 2023-05-17:
@dusty-nv I ran`apt-cache search deepstream` and the result just like you 
```
jet@jet-nano:~$ apt-cache search deepstream
deepstream-6.0 - Nvidia DeepStreamSDK runtime libraries, development files and samples
```
Does it means the DeepStream has been installed? Someone said I can located it from `/opt/nvidia/deepstream` and there are some samples I can try. But the `deepstream` folder doesn't exist. Would you mind to tell me where the installation path is? Or any recommendation for learning DeepStream?

dusty-nv, 2023-05-17:
@mmhzlrj try `sudo apt-get install deepstream-6.0`

I'm not an expert on DeepStream, so I would post your questions to https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/deepstream-sdk/15

If you prefer writing Python, I would check out these:   https://github.com/NVIDIA-AI-IOT/deepstream_python_apps

mmhzlrj, 2023-05-18:
Thank you @dusty-nv . After I running `sudo apt-get install deepstream-6.0` to install DeepStream-6.0, I can see the `deepstream` folder under `/opt/nvidia/`. And I can run `dpkg -l deepstream-6.0` to see more details:
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  deepstream-6.0 6.0.1-1      arm64        Nvidia DeepStreamSDK runtime libr
```
Thanks for your recommendation, I will look it up.

closed 2023-05-18 by mmhzlrj