dusty-nv/jetson-inference #1647: Linking lib jetson utils to independent project

LuisTbx, 2023-05-17:
Hi Dusty,

I would like to link and use both jetson-inference and jetson-utils in a self contained project. 

I have compiled and installed both jetson-utils and jetson-inference (sudo make install). The installed detectNet works (i had to also do sudo make install jetson-utils otherwise there were some errors).

I started by taking the detectNet example and creating a CMakeList.txt with the following content:

```

cmake_minimum_required(VERSION 3.16)
project(detectnet_test)

find_package(jetson-utils REQUIRED)
find_package(jetson-inference REQUIRED)

file(GLOB detectnetTestSources *.cpp)

cuda_add_executable(${PROJECT_NAME} ${detectnetTestSources})

target_link_libraries(${PROJECT_NAME} 
    ${jetson-inference_LIBS}
    ${jetson-utils_LIBS}
)

message(INFO " ${PROJECT_NAME}")

include_directories(
    ${jetson-inference_INCLUDE_DIR}
    ${jetson-utils_INCLUDE_DIR}
)

target_link_directories(${PROJECT_NAME} 
    PRIVATE 
    ${jetson-inference_INCLUDE_DIR}
    ${jetson-utils_INCLUDE_DIR}
)

```

The code compiles but finds linking problems,

```
/usr/bin/ld: detectnet.cpp:(.text._ZN9tensorNet14PROFILER_QUERYE13profilerQuery.part.0[_ZN9tensorNet18PrintProfilerTimesEv]+0x1f9): undefined reference to `Log::mLevel'
```

I understand this is an issue with jetson-utils but i am having hard time understanding why it happens. Could you spot any potential issue with the CMake file?

Thanks in advance.
Best Luis,

LuisTbx, 2023-05-17:
Dusty,

I figured it out, it turns  that the values for ```  ${jetson-inference_LIBS}, ${jetson-utils_LIBS}, ${jetson-inference_INCLUDE_DIRS} and  ${jetson-utils_INCLUDE_DIRS} ``` are retruning empty. Once i hardcoded the paths to the .so and includes the problem is gone.

Best, Luis

dusty-nv, 2023-05-17:
Thanks @LuisTbx - glad you got it working.  There is also an example CMakeLists.txt for externally including/linking jetson-inference/jetson-utils in out-of-tree projects here:

https://github.com/dusty-nv/jetson-inference/blob/master/examples/my-recognition/CMakeLists.txt

I think the difference is that I do it like this:  `target_link_libraries(my-recognition jetson-inference)`

LuisTbx, 2023-07-09:
Thanks @dusty-nv this was indeed the solution. Best Luis.