dusty-nv/jetson-inference #1648: Function LoadNetwork definition

SantiaSantia, 2023-05-19:
Hello Dusty, and anyone could help

in the tensorNet.cpp file (jetson-inference/ c) there was the code blow:
```
// LoadNetwork
bool tensorNet::LoadNetwork( const char* prototxt_path, const char* model_path, const char* mean_path, 
					    const char* input_blob, const char* output_blob, uint32_t maxBatchSize,
					    precisionType precision, deviceType device, bool allowGPUFallback,
					    nvinfer1::IInt8Calibrator* calibrator, cudaStream_t stream )
{
	std::vector<std::string> outputs;
	outputs.push_back(output_blob);
	
	return LoadNetwork(prototxt_path, model_path, mean_path, input_blob, outputs, maxBatchSize, precision, device, allowGPUFallback );
}
```
I searched the entire project end up failed finding the definition of LoadNetwork function. Then I assuming it was defined in the environment file like CUDA toolkit, TensorRT .etc. 

Would you please help telling where is the LoadNetwork definition?
It would be more grateful if you can teach me the methods for navigating function definition like the LoadNetwork.

dusty-nv, 2023-05-19:
Hi @SantiaSantia, the implementation of that overload of `tensorNet::LoadNetwork()` can be found here:

https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/c/tensorNet.cpp#L1090

SantiaSantia, 2023-05-21:
Hi dusty
Thank you for pointing it out. I saw that part in the searching process and just ignored it. How careless I was.
Thank you for saving me the time!

closed 2023-05-21 by SantiaSantia