dusty-nv/jetson-inference #1650: posenet compilation

SantiaSantia, 2023-05-22:
Hello Dusty and anyone could help

I was looking for the [make code] compiling the examples like posenet, imagenet, etc.
In the jetson-inference/build/examples/posenet/CMakeFiles/posenet.dir/build.make I found the code blow:
```
aarch64/bin/posenet: examples/posenet/CMakeFiles/posenet.dir/posenet.cpp.o
aarch64/bin/posenet: examples/posenet/CMakeFiles/posenet.dir/build.make
aarch64/bin/posenet: /usr/local/cuda/lib64/libcudart_static.a
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/librt.so
aarch64/bin/posenet: aarch64/lib/libjetson-inference.so
aarch64/bin/posenet: aarch64/lib/libjetson-utils.so
aarch64/bin/posenet: /usr/local/cuda/lib64/libcudart_static.a
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/librt.so
aarch64/bin/posenet: /usr/local/cuda/lib64/libnppicc.so
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/libopencv_calib3d.so.4.5.4
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/libopencv_features2d.so.4.5.4
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/libopencv_flann.so.4.5.4
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.5.4
aarch64/bin/posenet: /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.5.4
aarch64/bin/posenet: /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2.2.7
aarch64/bin/posenet: examples/posenet/CMakeFiles/posenet.dir/link.txt
        @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --bold --progress-dir=/home/zeyu/jetson-inference/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_2) "Linking CXX executable ../../aarch64/bin/posenet"
        cd /home/zeyu/jetson-inference/build/examples/posenet && $(CMAKE_COMMAND) -E cmake_link_script CMakeFiles/posenet.dir/link.txt --verbose=$(VERBOSE)

```
However, it seems to be dependencies instead of compilation. 
Would you help pointing out the code compiling posenet.cpp to posenet?
Or any hint from the CMakeFile to navigate to the makefile which responsible for the actual final compilation?
I was new to cmake&make system, thank you for your patience.

dusty-nv, 2023-05-22:
@SantiaSantia if you want to see the make invocation, run `VERBOSE=1 make`

This is an example CMakeLists.txt if you are trying to build your own posenet program outside of jetson-inference:  https://github.com/dusty-nv/jetson-inference/blob/master/examples/my-recognition/CMakeLists.txt

SantiaSantia, 2023-05-22:
@dusty-nv hi Dusty, after running `VERBOSE=1 make`,  I found out the info blow:
`/usr/bin/c++   -std=c++11 -Wno-write-strings -Wno-deprecated-declarations  -rdynamic CMakeFiles/posenet.dir/posenet.cpp.o  -o ../../aarch64/bin/posenet   -L/usr/lib/aarch64-linux-gnu/tegra  -Wl,-rpath,/usr/lib/aarch64-linux-gnu/tegra:/home/admin/jetson-inference/build/aarch64/lib:/usr/local/cuda/lib64:/opt/nvidia/vpi2/lib/aarch64-linux-gnu: /usr/local/cuda/lib64/libcudart_static.a -lpthread -ldl -lrt ../../aarch64/lib/libjetson-inference.so ../../aarch64/lib/libjetson-utils.so /usr/local/cuda/lib64/libcudart_static.a -lpthread -ldl -lrt -lGL -lGLU -lGLEW -lgstreamer-1.0 -lgstapp-1.0 -lgstpbutils-1.0 -lgstwebrtc-1.0 -lgstsdp-1.0 -lgstrtspserver-1.0 -ljson-glib-1.0 -lsoup-2.4 /usr/local/cuda/lib64/libnppicc.so -lnvbuf_utils -lnvinfer -lnvinfer_plugin -lnvcaffe_parser -lnvonnxparser /usr/lib/aarch64-linux-gnu/libopencv_calib3d.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_features2d.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_flann.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.5.4 /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2.2.7 `
So is it reasonable to see the cmake-make system as preparing the complex command line above(regarding compilation of `posenet` executable)?
Then while organizing such a complex project, shall we focus more on the `CMakeLists.txt` than the `Makefiles`?

dusty-nv, 2023-05-22:
@SantiaSantia yes when using CMake, you typically would only work with the CMakeLists.txt and not look at the makefiles that it generates

Are you training your ONNX model with the PyTorch scripts from this project (train.py or train_ssd.py)?    jetson-inference is setup for handling the pre/post-processing for those.   I have not used custom-trained pose models with it (the pose models I support come from trt_pose).  The pose models have a significant amount of post-processing involved.

SantiaSantia, 2023-05-23:
@dusty-nv I am currently building an industry AI module demo with the main function as recognizing object direction:

- (similar with `example/posenet`). Except that the usage requires only angle result output (1,2,3...to 360 degree). I assume this might save a lot of post-processing progress. 
- also there is another way: tag each degree and the network would be very similar as `example/imageNet`. This would be the prior way i would try.

In the real usage it requires both onboard training and off-board training for further network update. So i am going to work on both the `PyTorch train.py scripts from jetson-inference` and importing trained ONNX model. 
My next step would be focusing on read-understand the CMakeLists.txt for supporting future work. Grateful here for pointing me the direction and I would post the custom-trained ONNX model implantation result later here for everyone's reference.

dusty-nv, 2023-05-23:
@SantiaSantia that sounds interesting - recently I had been playing around with online training for this:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/webrtc-recognizer.md

It uses pytorch to train image classifier from interactively-collected data and hot-loads the ONNX models into TensorRT at runtime for inference.  I ended up pulling bits out of train.py and directly into the app for better integration.

Good luck with your project!

SantiaSantia, 2023-05-24:
@dusty-nv Thank you for the information, the simultaneous inference and re-training in `webrtc-recognizer `looks exciting! 
    
Recalling to the issue title, conclusion here for anyone's potential future reference:

1. While compiling `example/poseNet`, the Makefile generated code blow then executed under jetson-inference/build:
```
/usr/bin/c++   -std=c++11 -Wno-write-strings -Wno-deprecated-declarations  -rdynamic CMakeFiles/posenet.dir/posenet.cpp.o  -o ../../aarch64/bin/posenet   -L/usr/lib/aarch64-linux-gnu/tegra  -Wl,-rpath,/usr/lib/aarch64-linux-gnu/tegra:/home/admin/jetson-inference/build/aarch64/lib:/usr/local/cuda/lib64:/opt/nvidia/vpi2/lib/aarch64-linux-gnu: /usr/local/cuda/lib64/libcudart_static.a -lpthread -ldl -lrt ../../aarch64/lib/libjetson-inference.so ../../aarch64/lib/libjetson-utils.so /usr/local/cuda/lib64/libcudart_static.a -lpthread -ldl -lrt -lGL -lGLU -lGLEW -lgstreamer-1.0 -lgstapp-1.0 -lgstpbutils-1.0 -lgstwebrtc-1.0 -lgstsdp-1.0 -lgstrtspserver-1.0 -ljson-glib-1.0 -lsoup-2.4 /usr/local/cuda/lib64/libnppicc.so -lnvbuf_utils -lnvinfer -lnvinfer_plugin -lnvcaffe_parser -lnvonnxparser /usr/lib/aarch64-linux-gnu/libopencv_calib3d.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_features2d.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_flann.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.5.4 /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.5.4 /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2.2.7
```
2. the command seems dynamically linking the poseNet binary to libs blow:
```
    -lpthread: This library provides support for multi-threading.
    -ldl: This library provides functions for dynamic loading of shared libraries.
    -lrt: This library provides real-time extensions for POSIX-compliant operating systems.
    -lGL: This library provides the OpenGL API.
    -lGLU: This library provides the OpenGL Utility Library.
    -lGLEW: This library provides an extension loading library for OpenGL.
    -lgstreamer-1.0: This library provides the GStreamer multimedia framework.
    -lgstapp-1.0: This library provides GStreamer application support.
    -lgstpbutils-1.0: This library provides GStreamer plugin base utils.
    -lgstwebrtc-1.0: This library provides GStreamer WebRTC support.
    -lgstsdp-1.0: This library provides GStreamer SDP support.
    -lgstrtspserver-1.0: This library provides GStreamer RTSP server support.
    -ljson-glib-1.0: This library provides JSON manipulation support.
    -lsoup-2.4: This library provides an HTTP client/server library.
    /usr/local/cuda/lib64/libnppicc.so: This library provides the NVIDIA Performance Primitives Image Color Conversion functionality.
    /usr/lib/aarch64-linux-gnu/libopencv_calib3d.so.4.5.4: This library provides OpenCV calibration functions.
    /usr/lib/aarch64-linux-gnu/libopencv_features2d.so.4.5.4: This library provides OpenCV feature detection and description functions.
    /usr/lib/aarch64-linux-gnu/libopencv_flann.so.4.5.4: This library provides OpenCV approximate nearest neighbor search functions.
    /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.5.4: This library provides OpenCV image processing functions.
    /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.5.4: This library provides the core functionality of OpenCV.
    /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2.2.7: This library provides the NVIDIA Vision Programming Interface functionality.
and static library : 
/usr/local/cuda/lib64/libcudart_static.a: This is a static library that provides the CUDA runtime functionality. It is statically linked into the executable.
(infered by ChatGPT)
```

closed 2023-05-24 by SantiaSantia