dusty-nv/jetson-inference #1651: hello, can i change the detectnet to change the camera resolution in v4l2 camera and print the frame size to the console using print?

EvaBarret1, 2023-05-22:
hello, can i change the detectnet to change the camera resolution in v4l2 camera and print the frame size to the console using print?

dusty-nv, 2023-05-22:
Hi @EvaBarret1, you can try setting the camera resolution with `--input-width=N` and `--input-height=N` options:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

EvaBarret1, 2023-05-25:
and which variable is responsible for the output of what is in 'input'

dusty-nv, 2023-05-25:
The output resolution is based on the input resolution, unless you manually add code to resize the input image (i.e. using `cudaResize()`, `cudaCrop()`, or similar)

EvaBarret1, 2023-05-26:
yes, I would like to know which variable in the "Detectnet" code is responsible for displaying this data
[image]

EvaBarret1, 2023-05-26:
i need
gst camera video options

dusty-nv, 2023-05-26:
Are you using C++ or Python?   You can call `videoSource.GetOptions()` and in Python it will return a dict of the options, and in C++ it will return you the videoOptions struct:  https://github.com/dusty-nv/jetson-utils/blob/2ded2cec9de416ecc9e9c5a69f7010593d86817d/video/videoOptions.h

There are also convenience functions in videoSource like `videoSource.GetWidth()`, `GetHeight()`, `GetFrameRate()`, ect:   https://github.com/dusty-nv/jetson-utils/blob/2ded2cec9de416ecc9e9c5a69f7010593d86817d/video/videoSource.h#L293

AhmedL99, 2024-02-06:
Hello @dusty-nv, 
I have an issue when trying to write options:
`camera = videoSource("/work/input_resized.mp4", options= {'CodecType': 'CODEC_CPU'})`
`display = videoOutput("file://work/output_py.mp4", options= {'CodecType': 'CODEC_CPU'})`
I got this error: 
` camera = videoSource("/work/input_resized.mp4", options= {'CodecType': 'CODEC_CPU'})
TypeError: 'options' is an invalid keyword argument for this function`

dusty-nv, 2024-02-06:
Hi @AhmedL99, it would seem like your jetson-inference code is outdated (from before `options` was added to the Python bindings), or if you are running a container that container needs rebuilt.   Also the correct syntax should be `options= {'codecType': 'cpu'}`

AhmedL99, 2024-02-06:
Thank you @dusty-nv  for your fast response! 
This is weird, I cloned the repo a week ago! I will try to rebuild the container then. 
What about the `--headless` flag, is it configurable in Python script?