dusty-nv/jetson-inference #1655: Detectnet Model Creation in digits

berryliumsec, 2023-05-25:
Hi @dusty-nv 

I am following the tutorial to train detectnet using a custom dataset in digits. In the line below, the guide says to copy the detectnet.prototxt. 
https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/docs/detectnet-training.md?plain=1#L115

My question is, do i need to customize the detectnet.prototxt in the following areas or can i use as is?:

https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/data/networks/detectnet.prototxt#L11

https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/data/networks/detectnet.prototxt#L22

https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/data/networks/detectnet.prototxt#L44

https://github.com/dusty-nv/jetson-inference/blob/e1ee5676bdffc68f18569612403bc147cf9ae585/data/networks/detectnet.prototxt#L33

Is there anything else i need to

dusty-nv, 2023-05-25:
@berryliumsec it has been years since I used DIGITS and that part of the tutorial is deprecated (I will remove it at some point), but no I don't recall that you need to edit detectnet.prototxt

You would get better model performance using PyTorch and train_ssd.py like in the updated tutorial:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md

The old caffe-based detectnet.prototxt architecture is limited to a single object class and has slower inferencing time.

berryliumsec, 2023-05-25:
Thanks @dusty-nv Ill use the new tutorial

closed 2023-05-25 by berryliumsec