dusty-nv/jetson-inference #1657: libnvinfer.so: undefined reference to `nvdla::IProfile::setUseSoftMaxOptz(bool)'

ElenaSkrebneva, 2023-05-29:
Hei,
I have another isuue with the "make comment. Before you've just helped me solve an issue with "libcudla" (https://github.com/dusty-nv/jetson-inference/issues/1653) and I tried to continue with the "make", but got another one: 
"/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setUseSoftMaxOptz(bool)'
/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IPoolingLayer::setPoolingPaddingInclusionType(nvdla::PoolingPaddingInclusionType)'"
[image]
Could you please try to help me solve this one, too?
I have Jetson Nano SUB 4G version and Jetpack is 5.1.1-b56

ElenaSkrebneva, 2023-05-31:
It has it, because I've installed it (originally there was no JetPack in this SD card), using official instructions on: https://developer.nvidia.com/embedded/jetpack and installation went through. I didn't know about this compatibility issue. I will try to reinstall JetPack then I guess to the older version.

dusty-nv, 2023-05-31:
@ElenaSkrebneva I think because you have the 3rd-party SUB model of Jetson Nano, maybe it came with different SD card image?   With a fresh SD card image for it, you should be able to do `sudo apt-get update && sudo apt-get install nvidia-jetpack` if it doesn't come with JetPack.