dusty-nv/jetson-inference #1659: definition of videoOutput::Render

SantiaSantia, 2023-06-01:
Hello Dusty

I am currently looking at the output rendering mechanism in the jetson-utils. Found `virtual bool Render( void* image, uint32_t width, uint32_t height, imageFormat format );` in the videoOutput.h

I failed finding the definition of the Render function after looking at `videoOutput.cpp`, `videoOptions.h`, and` glDisplay.h` .

Could you help pointing out the original definition of the` videoOutput::Render` function?

With many thanks.

dusty-nv, 2023-06-01:
Hi @SantiaSantia, here are links to the Render() implementations:

* https://github.com/dusty-nv/jetson-utils/blob/8aa096774b76d899719244aa96b751337669f9ab/video/videoOutput.cpp#L211
* https://github.com/dusty-nv/jetson-utils/blob/8aa096774b76d899719244aa96b751337669f9ab/display/glDisplay.cpp#L637
* https://github.com/dusty-nv/jetson-utils/blob/8aa096774b76d899719244aa96b751337669f9ab/codec/gstEncoder.cpp#L627

SantiaSantia, 2023-06-02:
@dusty-nv Hi Dusty
in the code:
[](https://github.com/dusty-nv/jetson-utils/blob/8aa096774b76d899719244aa96b751337669f9ab/video/videoOutput.cpp#L218)
which is` if( !mOutputs[n]->Render(image, width, height, format) )`. I assume this Render was calling `gstEncoder::Render` and the deeper function for frame presenting is the `RenderOnce(image, width, height, format, 0, 0);` , which is inside glDisplay. Is it correct?

dusty-nv, 2023-06-02:
@SantiaSantia yes, by default it will attempt to create a secondary glDisplay output that gets called like that, so that you see the video on-screen in addition to the encoded output (when there is one)

SantiaSantia, 2023-06-05:
@dusty-nv Hello Dusty
Thank you for your time. Make a conclusion here for others future reference:

- The classify loop in imagenet.cpp consisted with 3 major parts, `CAPTURE`, `CLASSIFY`, and `RENDER`.
- The output of `CLASSIFY `is an image with result labeled on, and was passed to RENDER part as a pointer `uchar3* image`.
- `videoOutput::Render()` -> `glDisplay::Render()` -> `glDisplay::RenderOnce()` -> `glDisplay::RenderImage()` ->`glTexture::Render()`. The current border I could touch is before `glTexture::Render()`. The gl component was assumed as a fundamental components serving the `jetson-utils.`

closed 2023-06-05 by SantiaSantia