dusty-nv/jetson-inference #1662: Resume checkpoint - Re-training SSD-Mobilenet

ugmqu, 2023-06-02:
The instructions on this [page](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md) are a bit unclear.

How do I resume checkpoint after aborting the training

ugmqu, 2023-06-06:
using
`python3 train_ssd.py --data=data/lego --model-dir=models/lego --batch-size=4 --epochs=30 --resume=mb1-ssd-Epoch-7-Loss-7.6544482707977295.pth`
I get
`Traceback (most recent call last):
  File "train_ssd.py", line 12, in <module>
    import torch
ModuleNotFoundError: No module named 'torch'
`

dusty-nv, 2023-06-08:
> Traceback (most recent call last): File "train_ssd.py", line 12, in <module> import torch ModuleNotFoundError: No module named 'torch'

@ugmqu do you only get this import error when using `--resume` ?    Do you have PyTorch installed?    Can you run `python3 -c 'import torch; print(torch.__version__)'`

ugmqu, 2023-06-30:
reflashed the jetson nano. It works now

closed 2023-06-30 by ugmqu

4boris, 2023-07-27:
How do you find the name of the checkpoint path?

exploremore691, 2024-05-06:
I got the same issue 
using
python3 train_ssd.py --data=data/lego --model-dir=models/lego --batch-size=4 --epochs=30 --resume=mb1-ssd-Epoch-7-Loss-7.6544482707977295.pth
I get
Traceback (most recent call last): File "train_ssd.py", line 12, in <module> import torch ModuleNotFoundError: No module named 'torch' 
 
and then to fix it but 

/Downloads/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=data/img --model-dir=models/img
Traceback (most recent call last):
  File "train_ssd.py", line 20, in <module>
    from vision.ssd.vgg_ssd import create_vgg_ssd
  File "/home/me/Downloads/jetson-inference/python/training/detection/ssd/vision/ssd/vgg_ssd.py", line 6, in <module>
    from .predictor import Predictor
  File "/home/me/Downloads/jetson-inference/python/training/detection/ssd/vision/ssd/predictor.py", line 4, in <module>
    from .data_preprocessing import PredictionTransform
  File "/home/me/Downloads/jetson-inference/python/training/detection/ssd/vision/ssd/data_preprocessing.py", line 1, in <module>
    from ..transforms.transforms import *
  File "/home/me/Downloads/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 5, in <module>
    from torchvision import transforms
ModuleNotFoundError: No module named 'torchvision'