dusty-nv/jetson-inference #1663: No output of running multi label classification for image tagging

marco-cheung, 2023-06-03:
Hi there, I am trying to run sample code of below using Jetson Orin Nano Developer Kit, but no outputs. Checked that my terminal is located in the aarch64/bin directory and I am able to run "./imagenet.py images/orange_0.jpg images/test/output_0.jpg" and see the output file as expected. 

I've also got my input file in the directory:

[image]

dusty-nv, 2023-06-04:
@marco-cheung hmm this is what I get from running that same command on Orin Nano with JetPack 5.1.1:

```bash
[image]  loaded 'images/object_1.jpg'  (500x386, 3 channels)
imagenet:  99.61% class #11 (dog)
```

It makes a valid output image of it recognizing the dog.  I did notice that your command has a trailing `"` (after `images/test/tagging_1.jpg"`), I wonder if that causes anything.   Can you post your console log from when you run it?

Also, does it work on any of the `"images/object_*.jpg"` test images for you?

dusty-nv, 2023-06-05:
> In your sample code, there is indeed a trailing **"**. Not sure if it is a typo? If yes, maybe I should submit a pull request for this?

Sorry about that, you're right!   Just fixed it in commit https://github.com/dusty-nv/jetson-inference/commit/2d5310773b98d78ea56eb8d5b39ad4b25fb5bb58

marco-cheung, 2023-06-05:
I just made my first time to create a pull request for this issue on Github.
And then come back and saw you fixed it. (what a shame on me)

closed 2023-06-05 by marco-cheung