dusty-nv/jetson-inference #1666: WebRTC via https not working

ugmqu, 2023-06-05:
After genereting the key like mentioned, I still cannot connect via https. 
Chrome says the connection took too long and aborts.

Is there a quick workaround

ugmqu, 2023-06-06:
I do
cd python/www/html
python3 app.py --detection

inkognito mode did not help. Connection takes too long.

in the console i only get gStreamer outputs "timeout occoured waiting for the next buffer"

Maybe the certificate is malformed? Is it sufficient to just delete the .pem files and regenerate them?

dusty-nv, 2023-06-06:
> Maybe the certificate is malformed? Is it sufficient to just delete the .pem files and regenerate them?

Yes, I would try that

> cd python/www/html
> python3 app.py --detection

Can you try manually setting `--ssl-key` and `--ssl-cert` like this, and then post the console log here?

```bash
python3 app.py --detection --ssl-key=/path/to/your/key.pem --ssl-cert=/path/to/your/cert.pem
```

> in the console i only get gStreamer outputs "timeout occoured waiting for the next buffer"

HTTPS is only needed to receive camera video from the browser's webcam (like your laptop or phone camera), and those timeout messages are the Jetson waiting for the browser webcam to connect.  But for the time being, if you have a USB/CSI camera directly connected to your Jetson, you can launch app.py with `--input=/dev/video0` or `--input=csi://0`

ugmqu, 2023-06-06:
regenerating the keys I noticed an error in loading the random number generator

`547918843024:error:2406F079:random number generator:RAND_load_file:Cannot open file:../crypto/rand/randfile.c:88:Filename=/home/user/.rnd`

I commented out the line  `RANDFILE = $ENV::HOME/.rnd` in `/etc/ssl/openssl.cnf` 
like suggested [here](https://stackoverflow.com/questions/63893662/cant-load-root-rnd-into-rng) and the error did not appear again.

then using
`python3 app.py --detection --ssl-key=/data/key.pem --ssl-cert=/data/cert.pem --input=/dev/video0`
results in the error:
```
[webrtc] failed to load SSL certificate, unable to use HTTPS
[webrtc] (Failed to open file “/data/key.pem”: No such file or directory)
[webrtc] failed to create WebRTC server on port 8554
[gstreamer] gstEncoder -- failed to create encoder engine
```

dusty-nv, 2023-06-06:
> ```
> python3 app.py --detection --ssl-key=/data/key.pem --ssl-cert=/data/cert.pem --input=/dev/video0
> [webrtc] (Failed to open file “/data/key.pem”: No such file or directory)
> ```

@ugmqu I don't think that `/data` is a valid path from the `/jetson-inference/python/www/html` directory where you are running this from.  Double-check the location of your key/cert .pem files - may you meant to run it like this:

```bash
python3 app.py --detection --ssl-key=/jetson-inference/data/key.pem --ssl-cert=/jetson-inference/data/cert.pem --input=/dev/video0
```

ugmqu, 2023-06-06:
I again tried incognito mode, which did not work.

Then I again deleted my browser history in chrome.
Then I tried to connect via https again without closing the docker container. 
After I click proceed anyway it loaded very fast and worked.

Maybe the jetson is just too slow in the first time and chrome thinks that it was too suspicious

closed 2023-06-06 by ugmqu

dusty-nv, 2023-06-06:
@ugmqu ok, glad that you were able to get it working!

Sometimes during development when I was switching apps or certificates, I found it helpful to use incognito mode or clear the cache, or have Chrome Developer Tools open (`Ctrl+Shift+I`) and then do a `Hard Reload` by right-clicking on the refresh button and selecting it from the drop-down.

I'm not sure why Chrome gets confused sometimes with HTTPS, but it seems related to the locally-hosted self-signed certificates (which unfortunately is just what it is)

It's also sometimes to debug with the built-in WebRTC server by running video-viewer/detectnet/ect and navigating your browser to port `8554` (the same port WebRTC is running on).   This is the lower-level webpage which displays the connection stats and such:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/webrtc-server.md