dusty-nv/jetson-inference #1675: export stream to external webrtc  server

whsasf, 2023-06-14:
hi dusty, thank you for building this repo, it really helps a lot .
Now, i tried to export video stream to external webrtc server using below command: 

python3 app.py --input=/dev/video0  --output=webrtc://192.168.1.129:8889/mystream --headless

the logs:

captured 725 frames from /dev/video0 => webrtc://192.168.1.129:8889 (1280 x 720)
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer message qos ==> decoder
captured 750 frames from /dev/video0 => webrtc://192.168.1.129:8889 (1280 x 720)

but it doesn't seems to work , the stream doesn't even arrive 192.168.1.129 .So does this repo support the way above ? Thanks.

dusty-nv, 2023-06-14:
@whsasf the webrtc server in jetson-inference/jetson-utils operates in "listener" mode where it hosts the stream and accepts incoming connection attempts (similar to RTSP), and from there it can send/receive video streams to clients.

It's not expected to put a remote IP address in the `webrtc://` URL, that IP address signifies which network adapter on the Jetson you want to bind to (for example, `webrtc://0.0.0.0:8889/mystream` or `webrtc://@:8889/mystream` binds to all network interfaces on your Jetson)

Can you external webrtc connect to other webrtc servers, should you provide it with the URL of your Jetson's stream?

whsasf, 2023-06-15:
thank you for the response, i think i misunderstand the webrtc theory, i will find another way to this. thanks again.

closed 2023-06-15 by whsasf