dusty-nv/jetson-inference #1676: Getting real world 3D coordinates from depth map and image coordinates

dige-mothership, 2023-06-17:
Hi @dusty-nv, I was wondering if you have a function to derive the x,y,z 3D real word coordinates from the depth map and image coordinates in python? Or an explanation on how to do it?

dige-mothership, 2023-06-17:
I think what I was looking for is how to convert depth maps to point clouds, I'll do some googling but any insight you might offer will be great

dusty-nv, 2023-06-17:
@dige-mothership I don't have it in Python, but you can find my C++ code for converting the depth map to 3D point cloud coordinates (given the camera's intrinsic calibration coefficients) here:

https://github.com/dusty-nv/jetson-inference/blob/c6602dd46fd9a5fd46934db8933cb54b18665bae/c/depthNet.cpp#L510

dusty-nv, 2023-06-17:
Also I have the `cudaPointCloud` class from jetson-utils that does the point cloud extraction in CUDA:   https://github.com/dusty-nv/jetson-utils/blob/f0bff5c502f9ac6b10aa2912f1324797df94bc2d/cuda/cudaPointCloud.h

It does not have Python bindings for it though.

dige-mothership, 2023-06-18:
@dusty-nv thank you, I'll take a look

closed 2023-06-18 by dige-mothership

dige-mothership, 2023-07-26:
Hi @dusty-nv I am trying to save the pointcloud information here, im not sure what im doing wrong here
`/*
 * Copyright (c) 2021, NVIDIA CORPORATION. All rights reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a
 * copy of this software and associated documentation files (the "Software"),
 * to deal in the Software without restriction, including without limitation
 * the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the
 * Software is furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
 * THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 * FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
 * DEALINGS IN THE SOFTWARE.
 */

#include "videoSource.h"
#include "videoOutput.h"

#include "cudaOverlay.h"
#include "cudaMappedMemory.h"

#include "depthNet.h"

#include <signal.h>

bool signal_recieved = false;

void sig_handler(int signo)
{
	if( signo == SIGINT )
	{
		printf("received SIGINT\n");
		signal_recieved = true;
	}
}

int usage()
{
	printf("usage: depthnet [--help] [--network NETWORK]\n");
	printf("                [--colormap COLORMAP] [--filter-mode MODE]\n");
	printf("                [--visualize VISUAL] [--depth-size SIZE]\n");
	printf("                input_URI [output_URI]\n\n");
	printf("Mono depth estimation on a video/image stream using depthNet DNN.\n\n");
	printf("See below for additional arguments that may not be shown above.\n\n");
	printf("optional arguments:\n");
	printf("  --help            show this help message and exit\n");
	printf("  --network=NETWORK pre-trained model to load (see below for options)\n");
	printf("  --visualize=VISUAL controls what is displayed (e.g. --visualize=input,depth)\n");
	printf("                     valid combinations are:  'input', 'depth' (comma-separated)\n");
	printf("  --depth-size=SIZE  scales the size of the depth map visualization, as a\n");
	printf("                     percentage of the input size (default is 1.0)\n");
	printf("  --filter-mode=MODE filtering mode used during visualization,\n");
	printf("                     options are:  'point' or 'linear' (default: 'linear')\n");
	printf("  --colormap=COLORMAP depth colormap (default is 'viridis-inverted')\n");
	printf("                      options are:  'inferno', 'inferno-inverted',\n");
	printf("                                    'magma', 'magma-inverted',\n");
	printf("                                    'parula', 'parula-inverted',\n");
	printf("                                    'plasma', 'plasma-inverted',\n");
	printf("                                    'turbo', 'turbo-inverted',\n");
	printf("                                    'viridis', 'viridis-inverted'\n\n");
	printf("positional arguments:\n");
	printf("    input_URI       resource URI of input stream  (see videoSource below)\n");
[... 190 lines cut ...]
		// wait for the GPU to finish		
		CUDA(cudaDeviceSynchronize());

		// print out timing info
		net->PrintProfilerTimes();
	}
	

	/*
	 * destroy resources
	 */
	LogVerbose("depthnet:  shutting down...\n");
	
	SAFE_DELETE(input);
	SAFE_DELETE(output);
	SAFE_DELETE(net);
	
	CUDA_FREE_HOST(imgDepth);
	CUDA_FREE_HOST(imgComposite);

	LogVerbose("depthnet:  shutdown complete.\n");
	return 0;
}

`

dige-mothership, 2023-07-26:
Here is the error i get:

`home/agent/jetson-inference/examples/depthnet/depthnet.cpp: In function ‘int main(int, char**)’:
/home/agent/jetson-inference/examples/depthnet/depthnet.cpp:251:114: error: no matching function for call to ‘depthNet::SavePointCloud(const char*&, pixelType*&, uint32_t, uint32_t, float2&, float2&)’
   net->SavePointCloud(pointCloudFile,imgInput, input->GetWidth(), input->GetHeight(), focalLength, principalPoint);
                                                                                                                  ^
In file included from /home/agent/jetson-inference/examples/depthnet/depthnet.cpp:29:0:
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:230:7: note: candidate: bool depthNet::SavePointCloud(const char*)
  bool SavePointCloud( const char* filename );
       ^~~~~~~~~~~~~~
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:230:7: note:   candidate expects 1 argument, 6 provided
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:236:7: note: candidate: bool depthNet::SavePointCloud(const char*, float*, uint32_t, uint32_t)
  bool SavePointCloud( const char* filename, float* rgba, uint32_t width, uint32_t height );
       ^~~~~~~~~~~~~~
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:236:7: note:   candidate expects 4 arguments, 6 provided
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:242:7: note: candidate: bool depthNet::SavePointCloud(const char*, float*, uint32_t, uint32_t, const float2&, const float2&)
  bool SavePointCloud( const char* filename, float* rgba, uint32_t width, uint32_t height,
       ^~~~~~~~~~~~~~
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:242:7: note:   no known conversion for argument 2 from ‘pixelType* {aka float4*}’ to ‘float*’
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:249:7: note: candidate: bool depthNet::SavePointCloud(const char*, float*, uint32_t, uint32_t, const float (*)[3])
  bool SavePointCloud( const char* filename, float* rgba, uint32_t width, uint32_t height,
       ^~~~~~~~~~~~~~
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:249:7: note:   candidate expects 5 arguments, 6 provided
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:256:7: note: candidate: bool depthNet::SavePointCloud(const char*, float*, uint32_t, uint32_t, const char*)
  bool SavePointCloud( const char* filename, float* rgba, uint32_t width, uint32_t height,
       ^~~~~~~~~~~~~~
/home/agent/jetson-inference/build/x86_64/include/jetson-inference/depthNet.h:256:7: note:   candidate expects 5 arguments, 6 provided
[ 82%] Built target segnet
make[2]: *** [examples/depthnet/CMakeFiles/depthnet.dir/build.make:76: examples/depthnet/CMakeFiles/depthnet.dir/depthnet.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:474: examples/depthnet/CMakeFiles/depthnet.dir/all] Error 2
`

dusty-nv, 2023-07-26:
@dige-mothership try casting your img pointer to `float*` like this:

```cpp
net->SavePointCloud(pointCloudFile, (float*)imgInput, input->GetWidth(), input->GetHeight(), focalLength, principalPoint);
```

dige-mothership, 2023-07-26:
@dusty-nv Solved!!! thanks a million

dige-mothership, 2023-07-26:
@dusty-nv one more thing, ive been getting this error:

`[TRT]    depthNet::SavePointCloud() -- failed to upsample depth field`

and i see its from here

`		if( !Visualize((void*)depthField, width, height, IMAGE_GRAY32F, COLORMAP_NONE, FILTER_LINEAR) )
		{
			printf(LOG_TRT "depthNet::SavePointCloud() -- failed to upsample depth field\n");
			printf(LOG_TRT "NOT Extracting point cloud\n");
			return false;
		}
`

Can you think of any reason why i may be getting it?

dusty-nv, 2023-07-26:
@dige-mothership it's been a long time since I used that code so I'm not sure, but was there any other errors printed out?

dige-mothership, 2023-07-26:
@dusty-nv This is all of it:

`[TRT]    depthNet::Visualize() -- unsupported image format (gray32f)
[TRT]          supported formats are:
[TRT]              * rgb8
[TRT]              * rgba8
[TRT]              * rgb32f
[TRT]              * rgba32f
[TRT]    depthNet::SavePointCloud() -- failed to upsample depth field
`

dige-mothership, 2023-07-26:
I wonder if i the problem is the image type that is being passed to `depthNet::SavePointCloud() `

dusty-nv, 2023-07-26:
@dige-mothership try commenting out this if statement:

https://github.com/dusty-nv/jetson-inference/blob/a444f0c987e5584ad21616737fe543cf7fb794d9/c/depthNet.cpp#L408

Then recompile/reinstall.   I traced it through and it appears that it should work, that if check was added later.

Also there is https://github.com/dusty-nv/jetson-inference/tree/master/tools/depth-viewer if you haven't tried running that yet

dige-mothership, 2023-07-26:
@dusty-nv perfect, i have successfully extracted the pointcloud, thank you