dusty-nv/jetson-inference #1677: Can NOT publish rtsp stream

CapBarbossa, 2023-06-19:
Hi, dusty-nv!
   Me again! The guy who said bad words and didn't regret it the other day. And I am having still the same attitude: If what you do is wrong, then you ARE wrong. That's the only thing that matters, and you should NOT blame the way people point it out for you.
   Here is my question today. It is about publishing rtsp. I pulled the latest repo with tag of `model-mirror-190618` and compile it from source in `x86_64` platform. Despite the numerous bugs and the lowly CMakeLists.txt written, I finally built it successfully. Thanks to the knowledge of cmake I've been learning in the last two months. 
   When I run the binary `video-viewer` using `video-viewer rtsp://admin:xxx@192.168.1.10:554 rtsp://@:8554/my_stream`, I can NOT play this rtsp stream with `ffplay rtsp://127.0.0.1:8554/my_stream`. I checked the rtsp server with `netstat -tnlp` and the port number 8554 is there. The address `rtsp://admin:xxx@192.168.1.10:554 ` is my IP camera, and it works just fine. The binary `video-viewer` itself works fine too if I did't specify an output explicitly, and it can display the content of camera smoothly, with ZERO delay, which I think you deserve a thumb up! To debug this issue, I did two things:
  1. Using --debug option
      The corresponding error information is:
       ```
       [gstreamer] GST_LEVEL_WARNING GstAppSrc basesrc
              ../libs/gst/base/gstbasesrc.c:3127  gst_base_src_loop
              error: Internal data stream error.
[gstreamer] GST_LEVEL_WARNING GstAppSrc basesrc
              ../libs/gst/base/gstbasesrc.c:3127  gst_base_src_loop
              error: streaming stopped, reason not-linked (-1)
[gstreamer] GST_LEVEL_WARNING   rtspmedia
              ../gst/rtsp-server/rtsp-media.c:3250  default_handle_message
              0x7f45cdf6c970: got error Internal data stream error. (../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:media-pipeline/GstPipeline:pipeline1/GstAppSrc:mysource:
streaming stopped, reason not-linked (-1))
[gstreamer] GST_LEVEL_WARNING   rtspmedia
              ../gst/rtsp-server/rtsp-media.c:3919  gst_rtsp_media_prepare
              media 0x7f45cdf6c970 was not unprepared
[gstreamer] GST_LEVEL_ERROR     rtspclient
              ../gst/rtsp-server/rtsp-client.c:1087  find_media
              client 0x7f45cdf6d260: can't prepare media
[gstreamer] GST_LEVEL_ERROR     rtspclient
              ../gst/rtsp-server/rtsp-client.c:3346  handle_describe_request
              client 0x7f45cdf6d260: no media
[gstreamer] gstEncoder -- appsrc signalling enough data
       ```
  2. Pulled another repo: gst-rtsp-server
      I pulled this repo and compiled it. The built binaries like `test-mp4`, `test-video` worked just fine and the rtsp server can listen on 8554 port and I can play that rtsp stream also. So, the related gstreamer rtsp server libraries should be OK. 
This is all stuff I can provide with this rtsp issue and I hope they can help you to locate the bug.
Also, I want to dive into gstreamer program/software. It seems that you are good at this field, care to give me some hint about mastering this software? Some links or PDF books will be appreciated! Do not worry about if I could make it, I am a master in Linux!

dusty-nv, 2023-06-19:
> [gstreamer] GST_LEVEL_WARNING rtspmedia
> ../gst/rtsp-server/rtsp-media.c:3250 default_handle_message
> 0x7f45cdf6c970: got error Internal data stream error. (../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:media-pipeline/GstPipeline:pipeline1/GstAppSrc:mysource:

@CapBarbossa can you provide the rest of the console log, like where it prints out the gstreamer pipelines that it creates near the beginning?   On the capture side, when you run it with RTSP output is it fetching frames from your IP camera?

> I pulled the latest repo with tag of `model-mirror-190618` and compile it from source in `x86_64` platform.

As you have found, x86 isn't a primary support target in this project and I haven't even tested RTSP output on x86.  

> Also, I want to dive into gstreamer program/software. It seems that you are good at this field, care to give me some hint about mastering this software?

I wish I could, alas it's basically just a hodge-podge of google searches and other people's examples that I find when looking for a particular topic.   The RTSP and WebRTC servers in gstreamer seem especially difficult...good luck!

CapBarbossa, 2023-06-20:
I forgot to paste the capture side log, BIG sorry! Here it is the complaing text when I run `ffplay rtsp://127.0.0.1:8554/my_stream`:
```
[rtsp @ 0x7f7794000cc0] method DESCRIBE failed: 503 Service Unavailable
rtsp://127.0.0.1:8554/my_stream: Server returned 5XX Server Error reply
```
With `gst-play-1.0 rtsp://127.0.0.1:6666/my_stream`:
```
Now playing rtsp://127.0.0.1:8554/my_stream
Pipeline is live.
ERROR Unhandled error for rtsp://127.0.0.1:8554/my_stream
ERROR debug information: ../gst/rtsp/gstrtspsrc.c(6696): gst_rtspsrc_send (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstRTSPSrc:source:
Service Unavailable (503)
Reached end of play list.
```
Seems it is NOT capturing frames on capture side.
Below is the log at the begining part in sending side:
```
0:00:00.049019340 405849 0x5651008a6600 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
[gstreamer] initialized gstreamer, version 1.18.4.0
[gstreamer] gstDecoder -- creating decoder for admin:xxxx@192.168.1.64
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.1.64:554 latency=10 ! queue ! rtph264depay ! h264parse ! avdec_h264 name=decoder ! video/x-raw ! appsink name=mysink sync=false
[video]  created gstDecoder from rtsp://admin:xxxx@192.168.1.64:554
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:xxxx@192.168.1.64:554
     - protocol:  rtsp
     - location:  admin:xxxx@192.168.1.64
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      H264
  -- codecType:  cpu
  -- width:      1920
  -- height:     1080
  -- frameRate:  25
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       -1
  -- latency     10
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! x264enc name=encoder bitrate=4000 speed-preset=ultrafast tune=zerolatency key-int-max=30 insert-vui=1 ! video/x-h264 ! rtph264pay config-interval=1 name=pay0
[rtsp]   RTSP server started @ rtsp://parrot:6666
[video]  created gstEncoder from rtsp://127.0.0.1:8554/my_stream
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtsp://127.0.0.1:8554/my_stream
     - protocol:  rtsp
     - location:  127.0.0.1
     - port:      8554
  -- deviceType: ip
  -- ioType:     output
  -- codec:      H264
  -- codecType:  cpu
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- latency     10
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1600x900
[OpenGL] glDisplay -- X window resolution:    1600x900
[OpenGL] glDisplay -- display device initialized (1600x900)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      1600
  -- height:     900
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstBufferManager -- map buffer size was less than max size (3110400 vs 3110415)
[gstreamer] gstEncoder -- starting pipeline, transitioning to GST_STATE_PLAYING
[gstreamer] gstEncoder -- new caps: video/x-raw, width=1920, height=1080, format=(string)I420, framerate=30/1
```
And below snippet is the end part of the log when I terminated the `video-viewer`:
```
q[gstreamer] gstEncoder -- pipeline full, skipping frame 850 (1920x1080, 3110400 bytes)
video-viewer:  captured 850 frames (1920x1080)
^Creceived SIGINT
video-viewer:  shutting down...
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:6437  gst_rtsp_src_receive_response
              receive interrupted
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:6535  gst_rtspsrc_try_send
              receive interrupted
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:8932  gst_rtspsrc_pause
              PAUSE interrupted
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:6696  gst_rtspsrc_send
              error: Unhandled error
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:6696  gst_rtspsrc_send
              error: Option not supported (551)
[gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
              ../gst/rtsp/gstrtspsrc.c:8241  gst_rtspsrc_close
              error: Could not send message. (Generic error)
[gstreamer] gstDecoder -- pipeline stopped
[gstreamer] gstEncoder -- shutting down pipeline, sending EOS
[gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
[gstreamer] gstEncoder -- pipeline stopped
[rtsp]   RTSP server on port 8554 is shutting down
[rtsp]   waiting for RTSP server to stop...
[rtsp]   RTSP server thread stopped
video-viewer:  shutdown complete
```
Hope this can help. Since this repo isn't intended for X86_64, it's OK if this can't be solved. Nevertheless, I will go with the source C code and find the reason.

Fibo27, 2023-06-27:
Hi @CapBarbossa - i am not an expert on Gstreamer and have not used RTSP but have successfully used RTP - in my case I initially too had issues encoding and decoding - After multiple trials I realized that in my case h264 encoding was creating issues and so I switched to mjpeg. 
So to encode, I used port 1234 (any other too will work as long as they are allowed) and `gst-launch-1.0 udpsrc port=1234 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert ! autovideosink`. To decode I use
` ros2 launch ros_deep_learning detectnet.ros2.launch input_codec:=mjpeg input:=rtp://@:1234 output:=display://0 ` - I am using the ros_deep_learning repo but the key element is to ensure the 'input_codec:=mjpeg' followed by the port. FYI, my decoding is all on x86_64.

JIA-HONG-CHU, 2024-11-20:
Bro. Is there any update on this? tks

JIA-HONG-CHU, 2024-11-28:
> I forgot to paste the capture side log, BIG sorry! Here it is the complaing text when I run `ffplay rtsp://127.0.0.1:8554/my_stream`:
> 
> ```
> [rtsp @ 0x7f7794000cc0] method DESCRIBE failed: 503 Service Unavailable
> rtsp://127.0.0.1:8554/my_stream: Server returned 5XX Server Error reply
> ```
> 
> With `gst-play-1.0 rtsp://127.0.0.1:8554/my_stream`:
> 
> ```
> Now playing rtsp://127.0.0.1:8554/my_stream
> Pipeline is live.
> ERROR Unhandled error for rtsp://127.0.0.1:8554/my_stream
> ERROR debug information: ../gst/rtsp/gstrtspsrc.c(6696): gst_rtspsrc_send (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstRTSPSrc:source:
> Service Unavailable (503)
> Reached end of play list.
> ```
> 
> Seems it is NOT capturing frames on capture side. Below is the log at the begining part in sending side:
> 
> ```
> 0:00:00.049019340 405849 0x5651008a6600 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
> [gstreamer] initialized gstreamer, version 1.18.4.0
> [gstreamer] gstDecoder -- creating decoder for admin:xxxx@192.168.1.64
> [gstreamer] gstDecoder -- pipeline string:
> [gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.1.64:554 latency=10 ! queue ! rtph264depay ! h264parse ! avdec_h264 name=decoder ! video/x-raw ! appsink name=mysink sync=false
> [video]  created gstDecoder from rtsp://admin:xxxx@192.168.1.64:554
> ------------------------------------------------
> gstDecoder video options:
> ------------------------------------------------
>   -- URI: rtsp://admin:xxxx@192.168.1.64:554
>      - protocol:  rtsp
>      - location:  admin:xxxx@192.168.1.64
>      - port:      554
>   -- deviceType: ip
>   -- ioType:     input
>   -- codec:      H264
>   -- codecType:  cpu
>   -- width:      1920
>   -- height:     1080
>   -- frameRate:  25
>   -- numBuffers: 4
>   -- zeroCopy:   true
>   -- flipMethod: none
>   -- loop:       -1
>   -- latency     10
> ------------------------------------------------
> [gstreamer] gstEncoder -- codec not specified, defaulting to H.264
> [gstreamer] gstEncoder -- pipeline launch string:
> [gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! x264enc name=encoder bitrate=4000 speed-preset=ultrafast tune=zerolatency key-int-max=30 insert-vui=1 ! video/x-h264 ! rtph264pay config-interval=1 name=pay0
> [rtsp]   RTSP server started @ rtsp://parrot:6666
> [video]  created gstEncoder from rtsp://127.0.0.1:8554/my_stream
> ------------------------------------------------
> gstEncoder video options:
> ------------------------------------------------
>   -- URI: rtsp://127.0.0.1:8554/my_stream
>      - protocol:  rtsp
>      - location:  127.0.0.1
>      - port:      8554
>   -- deviceType: ip
>   -- ioType:     output
>   -- codec:      H264
>   -- codecType:  cpu
>   -- frameRate:  30
>   -- bitRate:    4000000
>   -- numBuffers: 4
>   -- zeroCopy:   true
>   -- latency     10
> ------------------------------------------------
> [OpenGL] glDisplay -- X screen 0 resolution:  1600x900
> [OpenGL] glDisplay -- X window resolution:    1600x900
> [OpenGL] glDisplay -- display device initialized (1600x900)
> [video]  created glDisplay from display://0
> ------------------------------------------------
> glDisplay video options:
> ------------------------------------------------
>   -- URI: display://0
>      - protocol:  display
>      - location:  0
>   -- deviceType: display
>   -- ioType:     output
>   -- width:      1600
>   -- height:     900
>   -- frameRate:  0
>   -- numBuffers: 4
>   -- zeroCopy:   true
> ------------------------------------------------
> [gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
> [gstreamer] gstBufferManager -- map buffer size was less than max size (3110400 vs 3110415)
> [gstreamer] gstEncoder -- starting pipeline, transitioning to GST_STATE_PLAYING
> [gstreamer] gstEncoder -- new caps: video/x-raw, width=1920, height=1080, format=(string)I420, framerate=30/1
> ```
> 
> And below snippet is the end part of the log when I terminated the `video-viewer`:
> 
> ```
> q[gstreamer] gstEncoder -- pipeline full, skipping frame 850 (1920x1080, 3110400 bytes)
> video-viewer:  captured 850 frames (1920x1080)
> ^Creceived SIGINT
> video-viewer:  shutting down...
> [gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:6437  gst_rtsp_src_receive_response
>               receive interrupted
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:6535  gst_rtspsrc_try_send
>               receive interrupted
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:8932  gst_rtspsrc_pause
>               PAUSE interrupted
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:6696  gst_rtspsrc_send
>               error: Unhandled error
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:6696  gst_rtspsrc_send
>               error: Option not supported (551)
> [gstreamer] GST_LEVEL_WARNING GstRTSPSrc rtspsrc
>               ../gst/rtsp/gstrtspsrc.c:8241  gst_rtspsrc_close
>               error: Could not send message. (Generic error)
> [gstreamer] gstDecoder -- pipeline stopped
> [gstreamer] gstEncoder -- shutting down pipeline, sending EOS
> [gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
> [gstreamer] gstEncoder -- pipeline stopped
> [rtsp]   RTSP server on port 8554 is shutting down
> [rtsp]   waiting for RTSP server to stop...
> [rtsp]   RTSP server thread stopped
> video-viewer:  shutdown complete
> ```
> 
> Hope this can help. Since this repo isn't intended for X86_64, it's OK if this can't be solved. Nevertheless, I will go with the source C code and find the reason.

Hello, did you solve this issue? Tks....

JIA-HONG-CHU, 2024-11-29:
Never mind. I solved it. Tks.