dusty-nv/jetson-inference #1679: Cuda errors and black screen when displaying a window

seb2189, 2023-06-20:
Regardless of trying `imagenet.py` or `video-viewer` the window stays black and cuda errors are shown. I'm running the docker container with the r32.7.1 version.    

`nvbuf_utils: dmabuf_fd 1083 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360`

Please find the complete log attached:
[video-viewer_error1.log](https://github.com/dusty-nv/jetson-inference/files/11801386/video-viewer_error1.log)

dusty-nv, 2023-06-20:
@seb2189 do you have a physical display attached to your Jetson?

If not, CUDA/OpenGL interoperability won't work over SSH with X11-forwarding.
If so, check `glxinfo` and `glxgears` (from the mesa-utils package) to confirm the NVIDIA graphics drivers are still installed.

When you are running headlessly, to view video remotely you should use WebRTC/RTP/RTSP streaming for the output:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#output-streams

BTW, if you want to avoid it trying to create an OpenGL window, run it with `--headless` flag.

seb2189, 2023-06-21:
Thanks for your response. Yes, I have a display attached and the glx commands are looking fine for me so far. Please find the log attached:
[glx.txt](https://github.com/dusty-nv/jetson-inference/files/11814083/glx.txt)

dusty-nv, 2023-06-21:
> `server glx vendor string: SGI`

From your glxinfo output, it is not using the NVIDIA graphics driver.  Can you try running the jetson-inference container to see if that has the same issue?   Or you may need to reinstall the jetpack graphics packages (or just reflash your SD card)

seb2189, 2023-06-21:
The log was already created from inside the container or what are you asking for? Here is the output from the container start up:
`ai-blox@tegra-ubuntu:~/workspace/jetson-inference$ docker/run.sh 
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.7.1
[sudo] password for ai-blox: 
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r32.7.1
DATA_VOLUME:      --volume /home/ai-blox/workspace/jetson-inference/data:/jetson-inference/data --volume /home/ai-blox/workspace/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/ai-blox/workspace/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/ai-blox/workspace/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/ai-blox/workspace/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/ai-blox/workspace/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
V4L2_DEVICES:     --device /dev/video0 
DISPLAY_DEVICE:   -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix `

dusty-nv, 2023-06-21:
Gotcha, okay - your OpenGL drivers are messed up outside of the container then too (on JetPack 4, the JetPack components are mounted into the containers)

You could try installing https://github.com/NVIDIA/libglvnd or the jetpack graphics packages - otherwise, I would re-flash your SD card.   glxinfo should show the graphics driver is `NVIDIA`

seb2189, 2023-06-22:
Building and installing libglvnd does not fix the issue or change the glxinfo output. Nevertheless, many thanks for your support.

closed 2023-06-23 by seb2189