dusty-nv/jetson-inference #1681: imagenet shutting down on USB camera with Jetson Nano

chesteryeum, 2023-06-21:
I'm running imagenet on a usb camera with a Jetson Nano (imagenet /dev/video0) and the process starts up but eventually shuts down after a minute or two. Either than a generic gstreamer capture error, there is no log item indicating why it failed. I've compiled a version with NVMM enabled and disabled, they both crash. Is there a way to capture more info on the failure?

*************************************************

imagenet:  12.79297% class #683 (oboe, hautboy, hautbois)

[TRT]    ------------------------------------------------
[TRT]    Timing Report /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   1.96798ms  CUDA   0.47776ms
[TRT]    Network       CPU  28.17573ms  CUDA  17.85932ms
[TRT]    Post-Process  CPU   0.28678ms  CUDA   0.26990ms
[TRT]    Total         CPU  30.43048ms  CUDA  18.60698ms
[TRT]    ------------------------------------------------

imagenet:  13.52539% class #758 (reel)

[TRT]    ------------------------------------------------
[TRT]    Timing Report /usr/local/bin/networks/Googlenet/bvlc_googlenet.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.09000ms  CUDA   0.46917ms
[TRT]    Network       CPU  26.60792ms  CUDA  14.33109ms
[TRT]    Post-Process  CPU   0.24193ms  CUDA   0.24120ms
[TRT]    Total         CPU  26.93986ms  CUDA  15.04146ms
[TRT]    ------------------------------------------------

[gstreamer] gstCamera::Capture() -- an error occurred retrieving the next image buffer
imagenet:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
imagenet:  shutdown complete.

dusty-nv, 2023-06-21:
Hi @chesteryeum, I have two suggestions for you:

1.  add some debug print statements to this function to find out where it is returning with an error (-1) :  https://github.com/dusty-nv/jetson-utils/blob/f0bff5c502f9ac6b10aa2912f1324797df94bc2d/codec/gstBufferManager.cpp#L299

2.  try changing this line from `break` to `continue` and see if the camera regains capturing:  https://github.com/dusty-nv/jetson-inference/blob/c6602dd46fd9a5fd46934db8933cb54b18665bae/examples/imagenet/imagenet.cpp#L147

chesteryeum, 2023-06-22:
@dusty-nv we added debug statements as per your suggestion and found that #2 solved the issue. Thanks for the quick support.

closed 2023-06-22 by chesteryeum