dusty-nv/jetson-inference #1684: Jetson Inference + Ros Deep Learning on x86

Fibo27, 2023-06-24:
@dusty-nv, in my earlier posts on your github channels, I have discussed building the jetson-inference & ros_deep_learning repos on x86 . 
The docker containers are available for various versions of JetPack and thank you for all the incredible work that you have done here. The changes that you have recently made to ensure that jetson-inference models work within the container and also the change to ensure that input-codec is included in the ros_deep_learning package is very much appreciated.

For those who would want to deploy jetson-inference and ros_deep_learning repos on x86 to take advantage of the higher compute capability, I can now say that this is possible to build from source and deploy it natively on the x86 hardware. If anyone is interested then the following may be of interest. (I haven't been able to build them yet on a docker container but that is project for later.)

- My system: x86 with 32 GB RAM + NVIDIA GPU +Ubuntu 22.04 + ROS2 Iron
- to Install Jetson Inference: Use these instructions: https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md
- make these changes here: https://github.com/dusty-nv/jetson-inference/blob/master/python/CMakeLists.txt and https://github.com/dusty-nv/jetson-utils/blob/f0bff5c502f9ac6b10aa2912f1324797df94bc2d/python/CMakeLists.txt
-
- `if(LSB_RELEASE_CODENAME MATCHES "focal")
	set(PYTHON_BINDING_VERSIONS **3.8**) #- change this to the Python version in your system - i had 3.10
else()
	set(PYTHON_BINDING_VERSIONS **2.7 3.6 3.7**) - you can keep them as is
endif()` 

- Note for x86, the python site-packages are stored in /opt/conda folder as shown below in these files https://github.com/dusty-nv/jetson-utils/blob/f0bff5c502f9ac6b10aa2912f1324797df94bc2d/python/bindings/CMakeLists.txt and https://github.com/dusty-nv/jetson-inference/blob/master/python/bindings/CMakeLists.txt
-
-`# on x86, install under /opt/conda/lib/pythonX.X/site-packages
# otherwise, install under /usr/lib/pythonX.X/dist-packages
if(CMAKE_SYSTEM_PROCESSOR MATCHES "x86_64")
	set(PYTHON_BINDING_INSTALL_DIR /opt/conda/lib/python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}/site-packages)`

- so make sure that the python path is correctly updated otherwise the python packages will not work - i am not using anaconda to manage my environments and so had to manually do this.

- For building ros_deep_learning node use the legacy install instructions. 
- Again the same issue related to the path come up as shown here https://github.com/dusty-nv/ros_deep_learning/blob/master/CMakeLists.txt
- `#locate dependencies
find_package(jetson-utils REQUIRED)
find_package(jetson-inference REQUIRED)` .

check https://stackoverflow.com/questions/28597351/how-do-i-add-a-library-path-in-cmake how to add the path.

I now have a robots (one on Rpi and another on a Jetson Nano) with Lidar, Sonar, IMU, Odometry sensors that also stream video on rtp (on a ros node0 and then I use the full compute power of my x86 to run rviz, the Detection algos of Ros Deep Learning, NAV2, SLAM, Robot Steering Packages of ROS2

@dusty-nv thank you again for the incredible work and more importantly your feedback. I haven't found the same level of feedback from the folks maintaining the Isaac-ROS repos. In fact, by using your repos, I don't really need the ones on Isaac ROS for my project.

dusty-nv, 2023-06-26:
@Fibo27 thank you so much for the comprehensive write-up of your changes, and work to get this running on x86.  I made a note about these modifications on my TODO list and will keep these in mind when maintaining the repo.   A lot of my issues with x86 are around supporting a wide variety of system environments as just one person (whereas with Jetson, most people are running JetPack), but the containers seem to help with that.  Good luck with your project and thanks again!  👍🤖

Fibo27, 2023-06-26:
Awesome

closed 2023-06-26 by Fibo27

ABDELRAHMAN123491, 2023-07-05:
Hi @Fibo27 
please have a look at [this issue](https://github.com/dusty-nv/jetson-inference/issues/1691), where I am attempting to build the apps in this repo on x86 architecture.

Fibo27, 2023-07-05:
Hi @ABDELRAHMAN123491 can you provide a detailed copy of the script - it is difficult to understand what exactly is causing the errors based on the details that you have provided

ABDELRAHMAN123491, 2023-07-06:
> Hi @ABDELRAHMAN123491 can you provide a detailed copy of the script - it is difficult to understand what exactly is causing the errors based on the details that you have provided

I guess the error is in building Gstreamer application. 
Here is the detailed script:
``
[jetson-inference]  Package selection status:  0
[jetson-inference]  No packages were selected for download.
 
[jetson-inference]  installation complete, exiting with status code 0
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/build
    $ ./install-pytorch.sh
 
[Pre-build]  Finished CMakePreBuild script
-- Finished installing dependencies
-- using patched FindCUDA.cmake
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr (found version "10.1") 
-- CUDA version: 10.1
-- Found OpenCV: /usr (found version "4.2.0") found components: core calib3d 
-- OpenCV version:  4.2.0
-- OpenCV version >= 3.0.0, enabling OpenCV
CMake Warning at CMakeLists.txt:106 (find_package):
  By not providing "FindVPI.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VPI", but
  CMake did not find one.

  Could not find a package configuration file provided by "VPI" (requested
  version 2.0) with any of the following names:

    VPIConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "VPI" to CMAKE_PREFIX_PATH or set "VPI_DIR"
  to a directory containing one of the above files.  If "VPI" provides a
  separate development package or SDK, be sure it has been installed.

-- didn't find VPI on system, disabling VPI
-- system arch:  x86_64
-- output path:  /home/wakeb/jetson-inference/build/x86_64
-- BUILD_EXPERIMENTAL disabled
-- Copying /home/wakeb/jetson-inference/c/actionNet.h
-- Copying /home/wakeb/jetson-inference/c/backgroundNet.h
-- Copying /home/wakeb/jetson-inference/c/calibration/randInt8Calibrator.h
-- Copying /home/wakeb/jetson-inference/c/depthNet.h
-- Copying /home/wakeb/jetson-inference/c/detectNet.h
-- Copying /home/wakeb/jetson-inference/c/imageNet.h
-- Copying /home/wakeb/jetson-inference/c/modelDownloader.h
-- Copying /home/wakeb/jetson-inference/c/poseNet.h
-- Copying /home/wakeb/jetson-inference/c/segNet.h
-- Copying /home/wakeb/jetson-inference/c/tensorConvert.h
-- Copying /home/wakeb/jetson-inference/c/tensorNet.h
-- Copying /home/wakeb/jetson-inference/c/tracking/objectTracker.h
-- Copying /home/wakeb/jetson-inference/c/tracking/objectTrackerIOU.h
-- Copying /home/wakeb/jetson-inference/c/tracking/objectTrackerKLT.h
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Deprecation Warning at tools/camera-capture/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[... 288 lines cut ...]
[ 99%] Building CXX object tools/camera-capture/CMakeFiles/camera-capture.dir/controlDetection.cpp.o
[ 99%] Building CXX object tools/camera-capture/CMakeFiles/camera-capture.dir/controlWindow.cpp.o
[ 99%] Built target segnet
[ 99%] Built target detectnet
[ 99%] Built target depthnet
[ 99%] Built target posenet
[ 99%] Built target actionnet
[ 99%] Built target jetson-inference-python-38
/home/wakeb/jetson-inference/tools/camera-capture/captureWindow.cpp: In member function ‘glWidget* CaptureWindow::GetWidget(int) const’:
/home/wakeb/jetson-inference/tools/camera-capture/captureWindow.cpp:222:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
[100%] Linking CXX executable ../../x86_64/bin/camera-capture
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_watch_send_messages'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_message_set_body_buffer'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_generate_digest_auth_response_from_md5'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/camera-capture/CMakeFiles/camera-capture.dir/build.make:192: x86_64/bin/camera-capture] Error 1
make[1]: *** [CMakeFiles/Makefile2:611: tools/camera-capture/CMakeFiles/camera-capture.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

``

Fibo27, 2023-07-06:
Seems like an issue with QT5 

See this thread and some similar ones

https://stackoverflow.com/questions/29455836/linkage-against-libqt5core

ABDELRAHMAN123491, 2023-07-11:
> Seems like an issue with QT5
> 
> See this thread and some similar ones
> 
> https://stackoverflow.com/questions/29455836/linkage-against-libqt5core

Thanks. I have solved this problem and manage to build the applications successfully. However, the C-applications work flawlessly; however, running any python file that imports jetson_utils causes an error as the jetson_utils module is not found. How could I solve such problem?

Thanks

stephemer, 2023-09-21:
_Make sure that the python path is correctly updated otherwise the python packages will not work - i am not using anaconda to manage my environments and so had to manually do this._

Do you mind sharing what path you updated to? I have a very similar setup but still getting "no module named jetson_inference" when attempting to run

ABDELRAHMAN123491, 2023-09-21:
> _Make sure that the python path is correctly updated otherwise the python packages will not work - i am not using anaconda to manage my environments and so had to manually do this._
>
> Do you mind [...]

You may write in the prompt ubuntu shell the following:
`which python` or `which python3`
The thing is that the path of the python environment is specific to the way you downloaded python.