dusty-nv/jetson-inference #1686: detectNet failed to load network

asmaaana, 2023-06-27:
when i run one of the detectnet models the below error is generated 
[TRT]    detected model format - custom  (extension '.')
[TRT]    model format 'custom' not supported by jetson-inference
[TRT]    detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network
Traceback (most recent call last):
  File "detectnet.py", line 55, in <module>
    net = detectNet(args.network, sys.argv, args.threshold)
Exception: jetson.inference -- detectNet failed to load network

dusty-nv, 2023-06-27:
Hi @asmaaana, what's the command-line that you are launching detectnet.py with?

asmaaana, 2023-06-27:
python3 detectnet.py --model=facedetect  /home/jetson/Desktop/jetson-inference/data/images/humans_0.jpg

dusty-nv, 2023-06-27:
@asmaaana that command works for me - I wonder if you need to update your jetson-inference repo?

Try this:

```bash
cd /home/jetson/Desktop/jetson-inference
git pull --recurse-submodules
cd build
cmake ../
make
sudo make install
```

dusty-nv, 2023-06-27:
Ah, try installing the updated dependencies again:

```
cmake -DBUILD_DEPS=YES ../
make
sudo make install
```

asmaaana, 2023-06-29:
It worked Thank you! i was able to run the above command successfully but then the output image will appear for less then .05 second and disappear is there a way to change this? as it disappears very quick and i can't even check the output

closed 2023-07-02 by asmaaana