dusty-nv/jetson-inference #1688: Resume training and add more classes

ugmqu, 2023-06-30:
Hi,

I resumend training multiple times, when I had more data. It worked very good.

But now I tried to use my already custom tained model and train it on data includig new classes.
I updated the labels.txt and the other txt val, train and test (I'm using pascal voc format).
But I get an error from pytorch about a size mismatch.

This is propably the mismatch of the number of classes. How do I add more classes?

dusty-nv, 2023-06-30:
@ugmqu if you had previously been using `--resume=CKPT`, try using `--pretrained-ssd=CKPT` instead.   Or just train it from the beginning on your entire new dataset with all the classes.

ugmqu, 2023-07-22:
'--pretrained-ssd=CKPT' works, but it somehow leads to a worse performance then training from scratch.

closed 2023-07-22 by ugmqu