dusty-nv/jetson-inference #1689: no member named ‘executeV2’; no member named ‘enqueueV2’

chancsc, 2023-07-02:
Just gotten my jetson nano, with ubuntu 18.04, Jetpack 4.2.1, python 3.6.9, TensorRT 5.1.6.1.

Encountered the error below while doing make, any advice? Thanks

[ 47%] Building CXX object CMakeFiles/jetson-inference.dir/c/poseNet.cpp.o
[ 48%] Building CXX object CMakeFiles/jetson-inference.dir/c/segNet.cpp.o
[ 49%] Building CXX object CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o
/home/nano/jetson-inference/c/tensorNet.cpp: In member function ‘bool tensorNet::ProcessNetwork(bool)’:
/home/nano/jetson-inference/c/tensorNet.cpp:1748:19: error: ‘class nvinfer1::IExecutionContext’ has no member named ‘executeV2’; did you mean ‘execute’?
    if( !mContext->executeV2(mBindings) )
                   ^~~~~~~~~
                   execute
/home/nano/jetson-inference/c/tensorNet.cpp:1756:19: error: ‘class nvinfer1::IExecutionContext’ has no member named ‘enqueueV2’; did you mean ‘enqueue’?
    if( !mContext->enqueueV2(mBindings, mStream, NULL) )
                   ^~~~~~~~~
                   enqueue
CMakeFiles/jetson-inference.dir/build.make:289: recipe for target 'CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2023-07-02:
@chancsc JetPack 4.2.1 is pretty old, I would either use an older branch of jetson-inference or upgrade your Nano to the latest JetPack 4 release from https://developer.nvidia.com/jetpack-sdk-464

chancsc, 2023-07-02:
thanks, that solved the problem.

closed 2023-07-02 by chancsc