dusty-nv/jetson-inference #1691: Run Some Functionalities on amd Architecture

ABDELRAHMAN123491, 2023-07-04:
Hi @dusty-nv 
Hope you are well. I would like to benefit from the bindings that you have provided in this repo in running the Gstreamer on an Amd64 architecture (normal Ubuntu-based PC), which has Nvidia 3090 GPU. Is that possible? or the work here only supports Jetson arch structure?

Thanks

ligaz, 2023-07-04:
[This issues](https://github.com/dusty-nv/jetson-inference/issues/1684) has a reference on running the library on x86. Hope you will find it useful.

ABDELRAHMAN123491, 2023-07-04:
> [This issues](https://github.com/dusty-nv/jetson-inference/issues/1684) has a reference on running the library on x86. Hope you will find it useful.

Thanks @ligaz 
I got the following error:
-- unsupported GNU version! gcc versions later than 8 are not supported!

ABDELRAHMAN123491, 2023-07-05:
I run into the following issue after attempting to run " make -j$(nproc)
":
```
usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_watch_send_messages'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_message_set_body_buffer'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Widgets.so.5.15.2: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgstrtspserver-1.0.so: undefined reference to `gst_rtsp_generate_digest_auth_response_from_md5'
/usr/bin/ld: /home/wakeb/anaconda3/lib/libQt5Core.so.5.15.2: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/camera-capture/CMakeFiles/camera-capture.dir/build.make:192: x86_64/bin/camera-capture] Error 1
make[1]: *** [CMakeFiles/Makefile2:611: tools/camera-capture/CMakeFiles/camera-capture.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
My system specifications are as follows:
`` L4T N.N.N [ JetPack UNKNOWN ]
   Ubuntu 20.04.6 LTS
   Kernel Version: 5.15.0-76-generic
 CUDA 11.4.48
   CUDA Architecture: NONE
 OpenCV version: 4.2.0
   OpenCV Cuda: NO
 CUDNN: 8.9.2.26
 TensorRT: 8.6.1.6
 Vision Works: NOT_INSTALLED
 VPI: NOT_INSTALLED
 Vulcan: NOT_INSTALLED
``
@dusty-nv @ligaz

ABDELRAHMAN123491, 2023-07-11:
I will close this issue as I have the solved the problem.

closed 2023-07-11 by ABDELRAHMAN123491