dusty-nv/jetson-inference #1693: Pytorch-SSD: onnx_export :The size of tensor a (12828) must match the size of tensor b (3000) at non-singleton dimension 1

SJavad, 2023-07-07:
Hi 🤗
I Have trained an SSD-Mobilenet object detection model with [pytorch-ssd](https://github.com/dusty-nv/pytorch-ssd) repository and it trained successfully with  --resolution=640 😊👍
but when I try to convert the .pth model to onnx model with onnx_export.py file with width and height 640, it gives me an error😕😔:
```python
    locations[..., :2] * center_variance * priors[..., 2:] + priors[..., :2],
RuntimeError: The size of tensor a (12828) must match the size of tensor b (3000) at non-singleton dimension 1
```
I try a lot of searching but no answer was found... 😩😡
this is the output of my command line:
```python
Namespace(net='ssd-mobilenet', input='', output='', labels='labels.txt', width=640, height=640, batch_size=1, model_dir='.\\models\\GED\\')
=> running on device cuda:0
=> found best checkpoint with loss 2.568583369255066 (.\models\GED\mb1-ssd-Epoch-6-Loss-2.568583369255066.pth)
creating network:  ssd-mobilenet
num classes:       3
=> loading checkpoint:  .\models\GED\mb1-ssd-Epoch-6-Loss-2.568583369255066.pth
=> exporting model to ONNX...
```
when I try to convert the model with default width and height (300 * 300) it is completely successful and converts...
I don't know why...

dusty-nv, 2023-07-07:
@SJavad does it work if you try 512x512 resolution?

also double-check that the labels.txt under your model directory doesn't have more than one `BACKROUND` class in it

SJavad, 2023-07-08:
> @SJavad does it work if you try 512x512 resolution?
> 
> also double-check that the labels.txt under your model directory doesn't have more than one `BACKROUND` class in it

@dusty-nv 
Thank you Dusty for your response 🖐
No, it doesn't work with 512*512 resolution: it can't convert to onnx, ERROR:
```python
locations[..., :2] * center_variance * priors[..., 2:] + priors[..., :2],
RuntimeError: The size of tensor a (8190) must match the size of tensor b (3000) at non-singleton dimension 1
```
also have one BACKGROUND in the label.txt file:
```text
BACKGROUND
object-1
object-goal
```

dusty-nv, 2023-07-08:
OK sorry, yes the onnx_export.py script needed updating - fixed it in https://github.com/dusty-nv/pytorch-ssd/commit/6accaa88845ec135a7d6fe25e9a26afd4698639d

I tested it here with `--resolution=512` - if you pull the latest jetson-inference or pytorch-ssd code and try exporting your model again, does it work for you?

SJavad, 2023-07-09:
> OK sorry, yes the onnx_export.py script needed updating - fixed it in [dusty-nv/pytorch-ssd@6accaa8](https://github.com/dusty-nv/pytorch-ssd/commit/6accaa88845ec135a7d6fe25e9a26afd4698639d)
> 
> I tested it here with `--resolution=512` - if you pull the latest jetson-inference or pytorch-ssd code and try exporting your model again, does it work for you?

@dusty-nv 
Thank you So much dusty 🖐🌸

closed 2023-07-09 by SJavad