dusty-nv/jetson-inference #1697: Does it need to checkout before compiling the project?

CapBarbossa, 2023-07-12:
Hi. Currently I am compiling this project from source. Just as the readme.md suggests:
```
git clone --recursive --depth=1 https://github.com/dusty-nv/jetson-inference
cd jetson-inference
mkdir build
cd build
cmake ../
make install
```
I did NOT run `git checkout `, but it ended compiling successfully. Problem is, when I run `import jetson.inference` in Python, it complains:
```
ImportError                               Traceback (most recent call last)
<ipython-input-2-d0bf3eeeabd4> in <module>
----> 1 import jetson.inference

/usr/lib/python3.6/dist-packages/jetson/inference/__init__.py in <module>
      3 
      4 # jetson.inference links against jetson.utils, and it needs loaded
----> 5 import jetson.utils
      6 
      7 # load jetson.inference extension module

/usr/lib/python3.6/dist-packages/jetson/utils/__init__.py in <module>
      2 #print("jetson.utils.__init__.py")
      3 
----> 4 from jetson_utils_python import *
      5 
      6 VERSION = '1.0.0'

ImportError: /usr/local/lib/python3.6/dist-packages/jetson_utils_python.so: undefined symbol: _Z12cudaDrawRectPvS_mm11imageFormatiiiiRK6float4
```
I googled for a while, and some `NVIDIA` guy said before compiling the project, I should checkout to specific version according to my jetson version, which in my case is `JetPack 4.6 [L4T 32.6.1]`, and he suggested I should check out to L4T-R32.7.1. But I browsered to this branch on github, finding the files were about 4 years ago! It does NOT make any sense, does it? This project is supposed to be an active one, no reason to use old versions right? So, what's the relation between `master` branch and those specific version branches? Should I checkout to a specific branch on my specific Jetson device? If the answer is no, then how to solve this `undefined symbol: _Z12cudaDrawRectPvS_mm11imageFormatiiiiRK6float4` bug?

dusty-nv, 2023-07-12:
> But I browsered to this branch on github, finding the files were about 4 years ago! It does NOT make any sense, does it? This project is supposed to be an active one, no reason to use old versions right? So, what's the relation between `master` branch and those specific version branches?

Those branches are for archiving purposes to "freeze" the code at the time of release of older JetPack versions that I may no longer build/test on myself.  For example, I no longer actively build on older than L4T R32.7.1 (JetPack 4.6.1).   However, for all intensive purposes the code should still build/run on older JetPack releases, and I doubt that your issue is related.

> ImportError: /usr/local/lib/python3.6/dist-packages/jetson_utils_python.so: undefined symbol: _Z12cudaDrawRectPvS_mm11imageFormatiiiiRK6float4

Are you able to start video-viewer and video-viewer.py?   Do both of those give same error?

Can you try doing `make && sudo make install` instead of the way you did it?

Do you still get this problem running the jetson-inference docker container?

CapBarbossa, 2023-07-13:
Thanks for replying.
As you mentioned, the built binary `video-viewer` does work smoothly. Bu not for `import jetson.inference` in Python. 
The corresponding python library is distributed to `/usr/local/` directory by `make install` as normal.

CapBarbossa, 2023-07-13:
And sorry to say that, I still can't understand the point: Can the main branch of the project be built and used in ` L4T R32.7.1 (JetPack 4.6.1)`? Both for binaries and python libraries.
By freeze code, I can get it. But I want to use the functions provided more recently and they are supposed to be accessible for  L4T R32.7.1 (JetPack 4.6.1), right?

dusty-nv, 2023-07-13:
Yes, I maintain jetson-inference master branch on L4T R32.7.x since that's a long-lived version.  I just double-checked and built it from scratch and it imported into Python fine.  Try this from your build dir:

```
sudo rm /usr/lib/python3.6/dist-packages/jetson_*.so
sudo make install 
sudo ldconfig
python3 -c 'import jetson_inference'
```

or check if the container works.