dusty-nv/jetson-inference #1699: Retraining ActionNet

dige-mothership, 2023-07-14:
Hi @dusty-nv do you know where I can find the pre trained model (in .pth format) for actionnet so I can train my own dataset? Also are there any instructions anywhere that I can follow? Thanks in advance

berryliumsec, 2023-07-14:
For context, i have taken a look at this https://developer.nvidia.com/blog/developing-and-deploying-your-custom-action-recognition-application-without-any-ai-expertise-using-tao-and-deepstream/ and it deploys using deepstream. I was wondering if there was a similar guide for training and deploying using jetson inference or a few pointers

dusty-nv, 2023-07-14:
@dige-mothership @berryliumsec I haven't trained these models, I believe the PTH checkpoints are from here:  https://github.com/kenshohara/3D-ResNets-PyTorch

The ONNX models I got from https://github.com/kn1ghtf1re/Activity-Recognition-TensorRT

dige-mothership, 2023-07-14:
@dusty-nv thanks for your always timely responses, really appreciate it. Ill take a look

closed 2023-07-14 by dige-mothership