dusty-nv/jetson-inference #1700: Compilation error and broken packages

Maddy1206, 2023-07-18:
Hello!

I have some compilation errors that are presumably related to the broken packages. 
Here is the compilation error:

`In file included from /home/unitree/jetson-inference/utils/codec/gstWebRTC.h:26:0,
                 from /home/unitree/jetson-inference/utils/codec/gstDecoder.cpp:24:
/home/unitree/jetson-inference/build/aarch64/include/jetson-utils/WebRTCServer.h:26:10: fatal error: libsoup/soup.h: No such file or directory
 #include <libsoup/soup.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/jetson-utils.dir/build.make:373: recipe for target 'utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/unitree/jetson-inference/utils/codec/gstWebRTC.h:26:0,
                 from /home/unitree/jetson-inference/utils/codec/gstEncoder.cpp:24:
/home/unitree/jetson-inference/build/aarch64/include/jetson-utils/WebRTCServer.h:26:10: fatal error: libsoup/soup.h: No such file or directory
 #include <libsoup/soup.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/jetson-utils.dir/build.make:397: recipe for target 'utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o] Error 1
CMakeFiles/Makefile2:703: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
`

and further above, while using `cmake -DBUILD_DEPS=YES ..`, it gives me an dependency issue with `libsoup2.4-dev`

`unitree@nx:~/jetson-inference/build$ cmake -DBUILD_DEPS=YES ..
 
Note:  this project uses git submodules in the source tree.
       if you haven't already, run the following command from
       the project's root directory:
 
           git submodule update --init

-- distro ID:       Ubuntu
-- distro version:  18.04
-- distro codename: bionic
-- Launching pre-build dependency installer script...
-- Build interactive:  YES
[Pre-build]  dependency installer script running...
[Pre-build]  build root directory: /home/unitree/jetson-inference/build
[Pre-build]  build interactive:    YES
[Pre-build]  build container:      NO
 
[sudo] password for unitree: 
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:2 http://ppa.launchpad.net/ubuntu-desktop/ubuntu-make/ubuntu bionic InRelease
Get:3 http://packages.microsoft.com/repos/code stable InRelease [3,569 B]
Hit:4 https://deb.nodesource.com/node_12.x bionic InRelease
Hit:5 https://repo.download.nvidia.com/jetson/common r32.5 InRelease
Hit:6 http://packages.ros.org/ros/ubuntu bionic InRelease
Get:7 https://repo.download.nvidia.com/jetson/t194 r32.5 InRelease [2,565 B]
Hit:8 https://librealsense.intel.com/Debian/apt-repo bionic InRelease
Get:9 http://packages.microsoft.com/repos/code stable/main arm64 Packages [73.6 kB]
Fetched 79.7 kB in 2s (38.7 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
dialog is already the newest version (1.3-20171209-1).
gstreamer1.0-libav is already the newest version (1.14.5-0ubuntu1~18.04.1).
qtbase5-dev is already the newest version (5.9.5+dfsg-0ubuntu2.6).
qtbase5-dev set to manually installed.
libglib2.0-dev is already the newest version (2.56.4-0ubuntu0.18.04.6).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.5).
libgstreamer1.0-dev is already the newest version (1.14.5-0ubuntu1~18.04.2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjson-glib-dev : Depends: libjson-glib-1.0-0 (= 1.4.2-3) but 1.4.2-3ubuntu0.18.04.1 is to be installed
                    Depends: gir1.2-json-1.0 (= 1.4.2-3) but 1.4.2-3ubuntu0.18.04.1 is to be installed
 libsoup2.4-dev : Depends: libsoup2.4-1 (= 2.62.1-1) but 2.62.1-1ubuntu0.4 is to be installed
                  Depends: gir1.2-soup-2.4 (= 2.62.1-1) but 2.62.1-1ubuntu0.4 is to be installed
E: Unable to correct problems, you have held broken packages.
`
Does anybody know what the issue here is?

dusty-nv, 2023-07-18:
> Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
> Hit:2 http://ppa.launchpad.net/ubuntu-desktop/ubuntu-make/ubuntu bionic InRelease
> Get:3 http://packages.microsoft.com/repos/code stable [...]

Hi @Maddy1206, it appears that you have added a bunch of other apt repos to your environment, I'm not sure why you can't `apt-get install libsoup2.4-dev`, but my guess is those other repos are related to the issue.

dusty-nv, 2023-07-18:
For comparison, these are the repos that `apt-get update` show on my JetPack 4.6.1 / L4T R32.7 system:

```bash
Hit:1 https://repo.download.nvidia.com/jetson/common r32.7 InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:3 https://repo.download.nvidia.com/jetson/t194 r32.7 InRelease
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [83.3 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [290 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [298 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main arm64 DEP-11 Metadata [5,376 B]
Get:10 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [10.0 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [70.9 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [57.3 kB]
```

You could try building the jetson-inference docker container if you want to use a clean environment for it.

Maddy1206, 2023-07-18:
I copied `/etc/apt/sources.list.d` into a backup folder with all files and put only the `nvidia-l4t-apt-source.list` back into the original folder.

Now the `apt-get update` looks like this:

```
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:2 http://archive.canonical.com/ubuntu bionic InRelease                     
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease           
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease            
Hit:5 https://repo.download.nvidia.com/jetson/common r32.5 InRelease           
Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease          
Hit:7 https://repo.download.nvidia.com/jetson/t194 r32.5 InRelease   
``` 

And I could install libsoup with the `apt-get install libsoup2.4-dev`. 
The compilation of the jetson-inference also worked (including the examples).

So thanks a lot for the hint!

closed 2023-07-18 by Maddy1206