dusty-nv/jetson-inference #1701: posenet model support

silent-code, 2023-07-18:
Hi Dusty, it appears that posenet only supports onnx pose models. Does it support the old trt-pose .pth models or do those have to be converted somehow to onnx format? If so, how does one convert .pth to onnx? 

Thanks in advance!

dusty-nv, 2023-07-18:
@silent-code jetson-inference uses TensorRT and doesn't directly load any PyTorch .pth model checkpoints, it always loads ONNX (or legacy caffemodels or .uff)

The trt_pose models I converted to ONNX using these scripts:

* https://github.com/dusty-nv/trt_pose/blob/6fa5f5d36c9be4bdf314838d840a82dd1f2960d3/onnx_export.py
* https://github.com/dusty-nv/trt_pose/blob/6fa5f5d36c9be4bdf314838d840a82dd1f2960d3/onnx_export.sh

silent-code, 2023-07-20:
Perfect thanks!

closed 2023-07-20 by silent-code