dusty-nv/jetson-inference #1702: failed to load detectNet model

lilhoser, 2023-07-19:
I'm trying to load a TAO converted model called VehicleMakeNet (https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models/vehiclemakenet).  I convert the model as follows:

```
ilhoser@whiteoak-jetson:~/Downloads/files$ ./tao-converter resnet18_vehiclemakenet_pruned.etlt -k tlt_encode -d 3,224,224 -o predictions/Softmax -m 1 -w 4294967296 -c vehiclemakenet_int8.txt

[INFO] [MemUsageChange] Init CUDA: CPU +221, GPU +0, now: CPU 249, GPU 3863 (MiB)
[INFO] [MemUsageChange] Init builder kernel library: CPU +302, GPU +156, now: CPU 574, GPU 4040 (MiB)
[WARNING] The implicit batch dimension mode has been deprecated. Please create the network with NetworkDefinitionCreationFlag::kEXPLICIT_BATCH flag whenever possible.
[INFO] ---------- Layers Running on DLA ----------
[INFO] ---------- Layers Running on GPU ----------
[INFO] [GpuLayer] CONVOLUTION: conv1/convolution + bn_conv1/batchnorm/mul_1 + bn_conv1/batchnorm/add_1 + activation_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_1a_conv_1/convolution + block_1a_bn_1/batchnorm/mul_1 + block_1a_bn_1/batchnorm/add_1 + block_1a_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_1a_conv_shortcut/convolution + block_1a_bn_shortcut/batchnorm/mul_1 + block_1a_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_1a_conv_2/convolution + block_1a_bn_2/batchnorm/mul_1 + block_1a_bn_2/batchnorm/add_1 + add_1/add + block_1a_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_1b_conv_1/convolution + block_1b_bn_1/batchnorm/mul_1 + block_1b_bn_1/batchnorm/add_1 + block_1b_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_1b_conv_shortcut/convolution + block_1b_bn_shortcut/batchnorm/mul_1 + block_1b_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_1b_conv_2/convolution + block_1b_bn_2/batchnorm/mul_1 + block_1b_bn_2/batchnorm/add_1 + add_2/add + block_1b_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_2a_conv_1/convolution + block_2a_bn_1/batchnorm/mul_1 + block_2a_bn_1/batchnorm/add_1 + block_2a_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_2a_conv_shortcut/convolution + block_2a_bn_shortcut/batchnorm/mul_1 + block_2a_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_2a_conv_2/convolution + block_2a_bn_2/batchnorm/mul_1 + block_2a_bn_2/batchnorm/add_1 + add_3/add + block_2a_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_2b_conv_1/convolution + block_2b_bn_1/batchnorm/mul_1 + block_2b_bn_1/batchnorm/add_1 + block_2b_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_2b_conv_shortcut/convolution + block_2b_bn_shortcut/batchnorm/mul_1 + block_2b_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_2b_conv_2/convolution + block_2b_bn_2/batchnorm/mul_1 + block_2b_bn_2/batchnorm/add_1 + add_4/add + block_2b_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_3a_conv_1/convolution + block_3a_bn_1/batchnorm/mul_1 + block_3a_bn_1/batchnorm/add_1 + block_3a_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_3a_conv_shortcut/convolution + block_3a_bn_shortcut/batchnorm/mul_1 + block_3a_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_3a_conv_2/convolution + block_3a_bn_2/batchnorm/mul_1 + block_3a_bn_2/batchnorm/add_1 + add_5/add + block_3a_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_3b_conv_1/convolution + block_3b_bn_1/batchnorm/mul_1 + block_3b_bn_1/batchnorm/add_1 + block_3b_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_3b_conv_shortcut/convolution + block_3b_bn_shortcut/batchnorm/mul_1 + block_3b_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_3b_conv_2/convolution + block_3b_bn_2/batchnorm/mul_1 + block_3b_bn_2/batchnorm/add_1 + add_6/add + block_3b_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_4a_conv_1/convolution + block_4a_bn_1/batchnorm/mul_1 + block_4a_bn_1/batchnorm/add_1 + block_4a_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_4a_conv_shortcut/convolution + block_4a_bn_shortcut/batchnorm/mul_1 + block_4a_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_4a_conv_2/convolution + block_4a_bn_2/batchnorm/mul_1 + block_4a_bn_2/batchnorm/add_1 + add_7/add + block_4a_relu/Relu
[INFO] [GpuLayer] CONVOLUTION: block_4b_conv_1/convolution + block_4b_bn_1/batchnorm/mul_1 + block_4b_bn_1/batchnorm/add_1 + block_4b_relu_1/Relu
[INFO] [GpuLayer] CONVOLUTION: block_4b_conv_shortcut/convolution + block_4b_bn_shortcut/batchnorm/mul_1 + block_4b_bn_shortcut/batchnorm/add_1
[INFO] [GpuLayer] CONVOLUTION: block_4b_conv_2/convolution + block_4b_bn_2/batchnorm/mul_1 + block_4b_bn_2/batchnorm/add_1 + add_8/add + block_4b_relu/Relu
[INFO] [GpuLayer] POOLING: flatten/Reshape
[INFO] [GpuLayer] CONVOLUTION: predictions/MatMul + predictions/BiasAdd
[INFO] [GpuLayer] SOFTMAX: predictions/Softmax
[INFO] [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +534, GPU +620, now: CPU 1158, GPU 4686 (MiB)
[INFO] [MemUsageChange] Init cuDNN: CPU +83, GPU +222, now: CPU 1241, GPU 4908 (MiB)
[INFO] Local timing cache in use. Profiling results in this builder pass will not be stored.
[INFO] Total Activation Memory: 4301670912
[INFO] Detected 1 inputs and 1 output network tensors.
[INFO] Total Host Persistent Memory: 87136
[INFO] Total Device Persistent Memory: 0
[INFO] Total Scratch Memory: 0
[INFO] [MemUsageStats] Peak memory usage of TRT CPU/GPU memory allocators: CPU 22 MiB, GPU 754 MiB
[INFO] [BlockAssignment] Started assigning block shifts. This will take 29 steps to complete.
[INFO] [BlockAssignment] Algorithm ShiftNTopDown took 0.585394ms to assign 3 blocks to 29 nodes requiring 1913344 bytes.
[INFO] Total Activation Memory: 1913344
[INFO] (Sparsity) Layers eligible for sparse math:
[INFO] (Sparsity) TRT inference plan picked sparse implementation for layers:
[INFO] [MemUsageChange] TensorRT-managed allocation in building engine: CPU +16, GPU +32, now: CPU 16, GPU 32 (MiB)

```

I get this error in detectNet:

```
lilhoser@whiteoak-jetson:~/Downloads/jetson-inference/build/aarch64/bin$ ./detectnet --model=~/Downloads/files/resnet18_vehiclemakenet_pruned_v101.engine --labels=~/Downloads/files/labels.txt --input-blob=input_1 --output=predictions/Softmax rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK display://0
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for 192.168.1.2
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 24.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, pixel-aspect-ratio=(fraction)1/1, width=(int)1280, height=(int)720, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK latency=10 ! queue ! rtph264depay ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink sync=false
[... 81 lines cut ...]
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::PillarScatterPlugin version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::ProposalDynamic version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::ROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::SeqLen2Spatial version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::SplitGeLU version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::VoxelGeneratorPlugin version 1
[TRT]    detected model format - engine  (extension '.engine')
[TRT]    loading network plan from engine cache... 
[TRT]    failed to load engine cache from 
[TRT]    failed to load 
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model
```

dusty-nv, 2023-07-19:
Hi @lilhoser, a few things:

(1) I'm not sure that detectNet is able to find your .engine file, double-check the path
(2) you are using the wrong command-line options to specify the output layers, it should use `--output-bbox` and `--output-cvg` like shown here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-tao.md#importing-your-own-tao-detection-models
(3) I recommend testing the model on a static image the first run in case you have issues with your RTSP camera which could confuse what the actual issue is.  instead, get the model running in detectNet first, then try it with camera inputs.

lilhoser, 2023-07-19:
@dusty-nv Thanks for the reply! detectNet actually works on the RTSP with a local pre-trained model like trafficcamnet, so I know this works.

I fixed the path - it did not accept `~/` for some reason. After changing `--output-bbox` and `--output-cvg` per your suggestion, I am seeing this error:

```
lilhoser@whiteoak-jetson:~/Downloads/jetson-inference/build/aarch64/bin$ ./detectnet --model=/home/lilhoser/Downloads/files/resnet18_vehiclemakenet_pruned_v101.engine --labels=/home/lilhoser/Downloads/files/labels.txt --input-blob=input_1 --output-cvg=output_cov/Sigmoid --output-bbox=output_bbox/BiasAdd rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK display://0
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for 192.168.1.2

(detectnet:95548): GStreamer-WARNING **: 16:37:47.631: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibav.so': /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

(detectnet:95548): GStreamer-WARNING **: 16:37:47.653: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibav.so': /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 24.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, pixel-aspect-ratio=(fraction)1/1, width=(int)1280, height=(int)720, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK latency=10 ! queue ! rtph264depay ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink sync=false
[video]  created gstDecoder from rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK
     - protocol:  rtsp
     - location:  192.168.1.2
     - port:      7447
  -- deviceType: ip
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      1280
  -- height:     720
  -- frameRate:  24
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- latency     10
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  3840x2160
[OpenGL] glDisplay -- X window resolution:    3840x2160
[OpenGL] glDisplay -- display device initialized (3840x2160)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      3840
  -- height:     2160
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        /home/lilhoser/Downloads/files/resnet18_vehiclemakenet_pruned_v101.engine
          -- input_blob   'input_1'
          -- output_cvg   'output_cov/Sigmoid'
          -- output_bbox  'output_bbox/BiasAdd'
          -- mean_pixel   0.000000
          -- class_labels /home/lilhoser/Downloads/files/labels.txt
          -- class_colors NULL
          -- threshold    0.500000
[... 72 lines cut ...]
                -- dim #2  224
[TRT]       binding 1
                -- index   1
                -- name    'predictions/Softmax'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  3
                -- dim #0  20
                -- dim #1  1
                -- dim #2  1
[TRT]    
[TRT]    binding to input 0 input_1  binding index:  0
[TRT]    binding to input 0 input_1  dims (b=1 c=3 h=224 w=224) size=602112
[TRT]    3: Cannot find binding of given name: output_cov/Sigmoid
[TRT]    failed to find requested output layer output_cov/Sigmoid in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to load /home/lilhoser/Downloads/files/resnet18_vehiclemakenet_pruned_v101.engine
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model

```

Changing from the RTSP input to a local image produced the same error.

I appreciate the help!

dusty-nv, 2023-07-20:
> [TRT]    CUDA engine context initialized on device GPU:
> [TRT]       -- layers       30
> [TRT]       -- maxBatchSize 1
> [TRT]       -- deviceMemory 1913344
> [TRT]       -- bindings     2
> [TRT] [...]

@lilhoser I didn't realize before seeing this that vehiclemakenet is a classification model, not a detection model.   It won't work with detectNet.

closed 2023-07-20 by lilhoser

lilhoser, 2023-07-20:
FYI, this `imagenet` command works:

```
lilhoser@whiteoak-jetson:~/Downloads/jetson-inference/build/aarch64/bin$ ./imagenet --model=/home/lilhoser/Downloads/files/resnet18_vehiclemakenet_pruned_v101.engine --labels=/home/lilhoser/Downloads/files/labels.txt --input-blob=input_1 --output-blob=predictions/Softmax 
rtsp://192.168.1.2:7447/wPXqmXadtJmll8FK display://0
```
but the classification is bogus because the frames aren't passing through the object detection classifier first. The model card mentions using DeepStream to chain the two classifiers together, so I'm looking into that.