dusty-nv/jetson-inference #1704: torch size mismatch

ugmqu, 2023-07-21:
`size mismatch for classification_headers.5.weight: copying a param with shape torch.Size([60, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([54, 256, 3, 3]).
	size mismatch for classification_headers.5.bias: copying a param with shape torch.Size([60]) from checkpoint, the shape in current model is torch.Size([54]).`

Hi, i get multiple size mismatch after running:
`python3 onnx_export.py --model-dir=models/...`

The checkpoints seem to have a bigger size with 31,3MB
Other models I trained, which worked fine have the checkpint size 30,1MB

dusty-nv, 2023-07-21:
@ugmqu is this trained with `train.py` or `train_ssd.py`?    I think what happens is that the number of classes differs from what the model is trained on.  Make sure you are loading the right labels.txt

dusty-nv, 2023-07-21:
Check the labels.txt in your model's folder to make sure that it doesn't have two `BACKGROUND` classes

train_ssd.py takes the labels.txt from your dataset and adds `BACKGROUND` class to it, and saves the updated labels.txt to your model's folder.  If the files/directories get mixed up, you can end up with multiple `BACKGROUND` classes in there

ugmqu, 2023-07-21:
I found the problem
There was a enter at the end of the labels.txt in the data folder.
It works now

closed 2023-07-22 by ugmqu