dusty-nv/jetson-inference #1706: Cannot generate ssh keys

ugmqu, 2023-07-27:
Running 

'openssl req -x509 -newkey rsa:4096 -keyout key.pem -out cert.pem -sha256 -days 365 -nodes -subj '/CN=localhost''

results in the error:

`req: Unrecognized flag subj /CN
`
do I miss any dependencies?

dusty-nv, 2023-07-27:
Hi @ugmqu, hmm which version of JetPack-L4T are you running?   You can check this with `cat /etc/nv_tegra_release`

I just tried this again on both JetPack 4.6.1 and JetPack 5.1, and it worked fine.   My `openssl version` is:

```bash
OpenSSL 1.1.1  11 Sep 2018    # JetPack 4.6
OpenSSL 1.1.1f  31 Mar 2020   # JetPack 5.1
```

ugmqu, 2023-08-10:
Something was wrong.

Fixed with reflashing throung Nvidia SDK Manager

closed 2023-08-10 by ugmqu